moveit/moveit2 #1782: No Planning Library Loaded

ranjitkathiriya, 2022-12-05:
Hello,

I am using Moveit2, and I am running the same launch file, mostly I am getting the error for **No Planning Library Loaded**, sometimes I am also able to load the library successfully and I am able to move my robot also.  I am attaching both screens below. 

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source  build
* humble branch

[Log link.](https://github.com/ranjitkathiriya/bug/blob/main/moveit.log)

[image]
[image]

MattiaDeiRossi, 2022-12-06:
Hi all,

I have the same issue on the same environment setup.
Is there a solution?

ranjitkathiriya, 2022-12-06:
Yes, I am also getting the same error too.

ilidar, 2022-12-27:
Hey everyone! Any updates on this one? This issue is crucial for my team, I would really appreciate someone looking at it.

Also, It's hard to reproduce with an Intel GPU and easy reproducable with NVIDIA graphics card.

mrmorten, 2023-01-17:
See issue #1049 - Character encoding UTF-8 issue / locale change to en_US.UTF-8 - it did the trick for me. 

On a fresh binary install of Humble and ROS + MoveIt with universal robots driver, same issues.

During testing after installation (without MoveIt) I noticed that only join_trajectory controller was the working one. forward and scaled joint did not work, not sure if related to same issue (UTF8 encoding).

Log file differences on working virtual machine vs failing (Non-nvidia vs Nvidia) - otherwise Intel i7 CPUs on laptops - both using Virtualbox 7 and Ubuntu 22.04, clean new install, everything was done according to the manual.

Working vs non-working driver:
 - Non-working one, does load the the joint trajectory controller; and gives errors on command interface, interpolation 
 - While the working version has problems loading "joint_trajectory_contr" and died, the result is that is actually working.

```
[ros2_control_node-1] [INFO] [1673972857.658538601] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ros2_control_node-1] [INFO] [1673972857.658549919] [scaled_joint_trajectory_controller]: Using 'splines' interpolation method.
```
Non-working RViz with MoveIt gives an extra error:

```
> [rviz2-2] [ERROR] [1673878337.416180088] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.ur_manipulator.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.ur_manipulator.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.
```

Commands executed (according to MoveIt manual for Humble):

```bash
ros2 launch ur_robot_driver ur_control.launch.py ur_type:=ur5 robot_ip:=yyy.yyy.yyy.yyy use_fake_hardware:=true launch_rviz:=false initial_joint_controller:=joint_trajectory_controller
ros2 launch ur_moveit_config ur_moveit.launch.py ur_type:=ur5 launch_rviz:=true use_fake_hardware:=true
```

I reproduced the fist execution error by the working version by not copy & paste the command correctly. It was missing the last letters for contr"oller". Did not affect the non-working environment to add the wrong controller. Still No planning library loaded.

Under Context in the MotionPlanning plugin for MoveIt in RViz2 : "NO PLANNING LIBRARY LOADED". Other differences include Warehouse link being different, failing one links to a .sqlite file, while the working one links to the localhost 127.0.0.1 IP. The working version has 3 options for kinematic solvers.

This all didn't make sense, since we did the exact same setup on two different computers. Until it did.

**Solution found in #1049 upon Googling the RViz error. Testing with adding LC_NUMERIC=en_US.UTF-8 ros2 launch ... worked.**

```bash
inter@BachelorUR5:~$ locale
LANG=en_GB.UTF-8
LANGUAGE=en_GB:en
LC_CTYPE="en_GB.UTF-8"
LC_NUMERIC=en_DK.UTF-8
LC_TIME=en_DK.UTF-8
LC_COLLATE="en_GB.UTF-8"
LC_MONETARY=en_DK.UTF-8
LC_MESSAGES="en_GB.UTF-8"
LC_PAPER=en_DK.UTF-8
LC_NAME=en_DK.UTF-8
LC_ADDRESS=en_DK.UTF-8
LC_TELEPHONE=en_DK.UTF-8
LC_MEASUREMENT=en_DK.UTF-8
LC_IDENTIFICATION=en_DK.UTF-8
LC_ALL=
```

```bash
inter@BachelorUR5:~$ cat /etc/default/locale
#  File generated by update-locale
LANG="en_US"
LANGUAGE="en_US:"
LC_NUMERIC="de_DE.UTF-8"
LC_TIME="de_DE.UTF-8"
LC_MONETARY="de_DE.UTF-8"
LC_PAPER="de_DE.UTF-8"
LC_NAME="de_DE.UTF-8"
LC_ADDRESS="de_DE.UTF-8"
LC_TELEPHONE="de_DE.UTF-8"
LC_MEASUREMENT="de_DE.UTF-8"
LC_IDENTIFICATION="de_DE.UTF-8"
[... 55 lines cut ...]
LC_ALL=en_US.UTF-8
```
```bash
inter@BachelorUR5:~$ locale
LANG=en_US.UTF-8
LANGUAGE=en
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```

[Humble Official Documentation](https://docs.ros.org/en/humble/Installation/Ubuntu-Install-Debians.html#set-locale) says: >"However, it should be fine if you’re using a different UTF-8 supported locale.". 
I can confirm that it is NOT fine! Many people change locales for local keyboard to function properly and also want their local date format. Choosing GB, DE or DK UTF8 variant is the logical choice in Europe as the formatting is correct. UTF-8 should be standard no matter what kind of UTF-8 it is. But it isn't in ROS (or even ROS2). Why? Must be hardcoded to en_US somewhere, instead of generic UTF-8.

Documentation should be more specific about the issue. Please provide a little more info detailing that only en_US.UTF8 works, for global settings and must not be overwritten with a different type of language setting in Ubuntu (until that is possible), even though it is UTF8.

henningkayser, 2023-01-23:
@mrmorten thank you for your extensive comment, pointing out the issue. I think I've tracked it down to being caused by locale-dependent double parsing, which apparently is still quite common in ROS packages. The problem is not that the locale is hardcoded anywhere, it's just that locales using comma-separated decimals can't parse doubles with dot notation if the package is using a locale-dependent parser. I've opened a new issue to track affected packages here https://github.com/ros-planning/moveit2/issues/1882

henningkayser, 2023-03-10:
@ranjitkathiriya could you check if this has been fixed for you already? #1882 is still listing open TODOs, though.