moveit/moveit2 #1787: Fix moveit_msg spelling of Aquire to Acquire.

azalutsky, 2022-12-06:
**Is your feature request related to a problem? Please describe.**
It is a non-issue beyond OCD people being disappointed in a spelling.

**Describe the solution you'd like**
moveit::planning_interface::MoveItErrorCode::
        UNABLE_TO_AQUIRE_SENSOR_DATA -> moveit::planning_interface::MoveItErrorCode::
        UNABLE_TO_ACQUIRE_SENSOR_DATA

**Describe alternatives you've considered**
Changing the spelling of acquire in the dictionary to aquire.

**Additional context**
[image]

tylerjw, 2022-12-06:
If you provide a PR to both moveit_msgs and moveit2 itself with adding your pr to the moveit2.repos file here the PR can test your change. Then we can merge the moveit_msgs change and then the moveit2.repos change can be updated to point at the moveit_msgs repo after your change. At the next release, we'll release them together and users will get both changes at once.

I believe that would work and is hopefully not too much trouble. Good catch on the spelling.

sameergupta4873, 2022-12-11:
@tylerjw I would like to work on this issue, according to  the above discussion, I broke the issue into few steps:-

- I have to fix the typo in the moveit_msg and send a PR to that.
- in the PR to the moveit.repos, I have to update the PR version of moveit_msgs in the moveit.repos.
- so that it can be tested if it works or not, also in the moveit.repos last commit removed the moveit_msgs dependency, so 
I think adding it again wouldn't be a problem?
- also if the moveit_msgs test works fine, I can remove it from the moveit.repos again.

AtharvSabde, 2023-01-19:
@tylerjw I would like to work on this issue

closed 2025-01-29 by github-actions[bot]