moveit/moveit2 #1795: Replace empty classes with namespaces

ChrisThrasher, 2022-12-08:
**Describe the solution you'd like**
Convert classes like [`trajectory_processing::RuckigSmoothing`](https://github.com/ros-planning/moveit2/blob/main/moveit_core/trajectory_processing/include/moveit/trajectory_processing/ruckig_traj_smoothing.h#L45) with namespaces. Because they contain no non-static data members, there is no reason these need to be classes. All the private functions are better represented by free functions in an anonymous namespaces in a .cpp file where they're have internal linkage and don't require adding more headers to the interface of this namespace.

**Additional context**
This will not even be an API break for users unless they're for some reason creating an instance of this empty type so it's a safe change to make.

AndyZe, 2022-12-08:
The idea sounds OK to me.

tylerjw, 2022-12-08:
This sounds good to me, and I appreciate that you won't break the API doing this. :+1:

closed 2024-11-08 by ChrisThrasher