moveit/moveit2 #1811: Moveit2 doesnt work with simulation time

TheConstructAi, 2022-12-15:
### Description

We had to set the `use_sim_time` hardcoded her [MOD to launches.py](https://github.com/TheConstructAi/moveit2/blob/humble/moveit_configs_utils/moveit_configs_utils/launches.py#L232)
To avoid failing the execution of trajectories.

The question is how is it supposed to be set using the new system of : 

`def generate_launch_description():
    moveit_config = MoveItConfigsBuilder("panda", package_name="my_panda_moveit_config").to_moveit_configs()
    return generate_move_group_launch(moveit_config)`

We imagine that this has to be done somehow, but what is the way the moveit2 developers consider that it should be set? Because the setup assistant doesn't seem to consider that possibility in the config files anywhere when it generates the config files.

### Your environment
* ROS Distro: HUMBLE
* OS Version: Ubuntu 22.04
* Source or Binary build? SOurce
* If source, which branch? Humble, forked here: [moveit2](https://github.com/TheConstructAi/moveit2)

### Steps to reproduce
Launch a simulation with a movitconfig generated with wizard.
You can try it using this ROSject: https://app.theconstructsim.com/l/537b5a08/

### Expected behaviour
It should work with simulations by being able to set somewhere easy the use_sim_time, or detect that the clock topic is there.

### Actual behaviour
Unless you hardcode to use the ' use_sim_time'  the trajectories in moveitr are planned but not executed

henningkayser, 2022-12-19:
The config builder is just producing a dict in the end, so adding or overriding values after returning the config would be the way to go, currently. In this particular case it's debatable if adding support for a MoveGroup config file would be more adequate, though.

jeans-s12, 2023-09-14:
I have the same problem, I need to transfer "use_sim_time: true", how can I do this?

DiegoCarvajal98, 2023-10-01:
> I have the same problem, I need to transfer "use_sim_time: true", how can I do this?

I just set the use_sim_time parameter to true after starting everything and is working, the command I used is this one:
```
ros2 param set /move_group use_sim_time true
```

closed 2024-03-15 by github-actions[bot]