moveit/moveit2 #1814: Error while Demo launch -  on ROS2 Humble and 22.04 Ubuntu

ClaBirk, 2022-12-17:
Hello,

i am trying to use Moveit2 on ROS2 Humble for a Project at univerity.
First goal: get a 6DOF Robot running in RVIZ and Moveit2 
Installation of ROS2 and Moveit went fine as far as i can tell.
But i can´t see the robot in Rviz:

ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true
From the moveit_ros_visualization folder, choose “MotionPlanning” as the DisplayType. Press “Ok”.

-> No Robot visible 
I changed this: LC_NUMERIC=en_US.UTF-8 but still no Robot visible

This is the  Terminal output:

cb@cb-vb:~$ ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true

[INFO] [launch]: All log files can be found below /home/cb/.ros/log/2022-12-17-20-37-18-197913-cb-vb-14381

[INFO] [launch]: Default logging verbosity is set to INFO

[INFO] [rviz2-1]: process started with pid [14382]

[INFO] [static_transform_publisher-2]: process started with pid [14384]

[INFO] [robot_state_publisher-3]: process started with pid [14386]

[INFO] [move_group-4]: process started with pid [14388]

[INFO] [ros2_control_node-5]: process started with pid [14390]

[INFO] [ros2 run controller_manager spawner panda_arm_controller-6]: process started with pid [14392]

[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process started with pid [14395]

[INFO] [ros2 run controller_manager spawner joint_state_broadcaster-8]: process started with pid [14397]

[static_transform_publisher-2] [WARN] [1671305838.815661095] []: Old-style arguments are deprecated; see --help for new-style arguments

[rviz2-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

[static_transform_publisher-2] [INFO] [1671305838.861997555] [static_transform_publisher]: Spinning until stopped - publishing transform

[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')

[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')

[static_transform_publisher-2] from 'world' to 'panda_link0'

[move_group-4] [INFO] [1671305838.975413848] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0479865 seconds

[move_group-4] [INFO] [1671305838.975631262] [moveit_robot_model.robot_model]: Loading robot model 'panda'...

[ros2_control_node-5] [INFO] [1671305838.987486640] [resource_manager]: Loading hardware 'PandaFakeSystem' 

[ros2_control_node-5] [INFO] [1671305838.990608752] [resource_manager]: Initialize hardware 'PandaFakeSystem' 

[ros2_control_node-5] [WARN] [1671305838.990660318] [mock_generic_system]: Parsing of optional initial interface values failed or uses a deprecated format. Add initial values for every state interface in the ros2_control.xacro. For example: 

[ros2_control_node-5] <state_interface name="velocity"> 

[ros2_control_node-5]   <param name="initial_value">0.0</param> 

[ros2_control_node-5] </state_interface>

[ros2_control_node-5] [INFO] [1671305838.990669681] [resource_manager]: Successful initialization of hardware 'PandaFakeSystem'

[ros2_control_node-5] [INFO] [1671305838.990686914] [resource_manager]: Loading hardware 'PandaHandFakeSystem' 

[ros2_control_node-5] [INFO] [1671305838.990694737] [resource_manager]: Initialize hardware 'PandaHandFakeSystem' 
[... 419 lines cut ...]
[INFO] [ros2 run controller_manager spawner panda_hand_controller-7]: process has finished cleanly [pid 14395]

[rviz2-1] Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.

[rviz2-1]          at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp

[rviz2-1] [ERROR] [1671305853.382136582] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available

[rviz2-1] [INFO] [1671305853.414680474] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.

[rviz2-1] [INFO] [1671305853.494084358] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0540943 seconds

[rviz2-1] [INFO] [1671305853.494162475] [moveit_robot_model.robot_model]: Loading robot model 'panda'...

[rviz2-1] [ERROR] [1671305853.521849018] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.panda_arm.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.panda_arm.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.

[rviz2-1] [ERROR] [1671305917.881814249] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available

[rviz2-1] [INFO] [1671305917.899123944] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.

[rviz2-1] [INFO] [1671305917.971941664] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.0537786 seconds

[rviz2-1] [INFO] [1671305917.971977593] [moveit_robot_model.robot_model]: Loading robot model 'panda'...

[rviz2-1] [ERROR] [1671305917.980618104] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_kinematics.panda_arm.kinematics_solver_timeout' has invalid type: Wrong parameter type, parameter {robot_description_kinematics.panda_arm.kinematics_solver_timeout} is of type {double}, setting it to {string} is not allowed.

henningkayser, 2022-12-19:
The last line of your log is showing you the reason why the robot model couldn't be loaded: an invalid parameter. Please review your kinematics.yaml and fix the value type of `kinematics_solver_timeout`.

ClaBirk, 2022-12-20:
Seems like this is a consequential error. I checked it it is of type float.

sjahr, 2022-12-21:
@ClaBirk I hope that your problem is resolved. In case you run into further problems, you can also check out the [Troubleshooting issue ](https://github.com/ros-planning/moveit2/issues/1599) and if that does not help, re-open this issue or a new one

closed 2022-12-21 by sjahr

develiberta, 2023-08-01:
I'm having the same problem. Rviz opens well on my server, but the No Planning Library loaded. Especially, I think the problem is the log of this issue I have in common. It is as follows.
`[moveit.ros_planning.planning_pipeline]: Exception while loading planner 'chomp_interface/CHOMPPlanner': According to the loaded plugin descriptions the class chomp_interface/CHOMPPlanner with base class type planning_interface::PlannerManager does not exist. Declared types are  ompl_interface/OMPLPlanner pilz_industrial_motion_planner/CommandPlanner stomp_moveit/StompPlannerAvailable plugins: ompl_interface/OMPLPlanner, pilz_industrial_motion_planner/CommandPlanner, stomp_moveit/StompPlanner`

Mustafaalam99, 2024-03-23:
i'm having similar problem, in my case the robot model is not loaded.

[rviz2-3] [ERROR] [1711174126.096524814] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
[rviz2-3] [INFO] [1711174126.125103229] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
[rviz2-3] [ERROR] [1711174136.286465461] [rviz]: Could not find parameter robot_description_semantic and did not receive robot_description_semantic via std_msgs::msg::String subscription within 10.000000 seconds.
[rviz2-3] Error:   Could not parse the SRDF XML File. Error=XML_ERROR_EMPTY_DOCUMENT ErrorID=13 (0xd) Line number=0
[rviz2-3]          at line 732 in /home/mustafa/ws_moveit2/src/srdfdom/src/model.cpp
[rviz2-3] [ERROR] [1711174136.292062514] [moveit_rdf_loader.rdf_loader]: Unable to parse SRDF
[rviz2-3] [ERROR] [1711174136.321760382] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Robot model not loaded

This is what my terminal gives. Any help would be appreciated.

Thanks

zhuxiyu24, 2024-11-07:
俺也一样

BengIslam7, 2025-02-06:
@Mustafaalam99 Did you solve this problem because i have the same ?