moveit/moveit2 #1824: Do a circle with a real robot

marioesteban11, 2022-12-21:
Hi everyone,

Im in ROS2 Humble, in ubuntu 22.04.

I'm using a ur3e robot with this moveit repository, and I was wondering if anyones knows/has an idea of how to do a circle using moveit and RVIZ simulator.

Thanks in advance.

mrceki, 2023-01-02:
You can do using moveit servo".

marioesteban11, 2023-01-03:
Do you have any repo or webpage where I can look for it? I didn't find nothing like that

mrceki, 2023-01-03:
https://ros-planning.github.io/moveit_tutorials/doc/realtime_servo/realtime_servo_tutorial.html
https://moveit.picknik.ai/humble/doc/examples/realtime_servo/realtime_servo_tutorial.html

sjahr, 2023-04-13:
Closing this due to inactivity. @marioesteban11 Please re-open if the question is still open

closed 2023-04-13 by sjahr