moveit/moveit2 #1829: Movegroup with python

marioesteban11, 2022-12-23:
Hi everyone,

I'm using ROS2 Humble in ubuntu 22.04.

Im tryíng to do an easy program with move group interface in python, but I don't really know how to do it.

I'm trying to move an Universal robot system with this interface but I have no idea how to do it.

Do you have any tutorial that could help me, or any idea?

Thanks in advance.

henningkayser, 2023-01-20:
There are currently no supported Python interfaces in MoveIt 2. However, there is an open Pull Request that is adding Python bindings for MoveItCpp here https://github.com/ros-planning/moveit2/pull/1546 with tutorials being added here https://github.com/ros-planning/moveit2_tutorials/pull/565, https://github.com/ros-planning/moveit2_tutorials/pull/564. Reviews and testers are always welcome ;)

JafarAbdi, 2023-04-12:
The mentioned PRs to support python binding are in the latest main. See https://moveit.picknik.ai/main/doc/examples/motion_planning_python_api/motion_planning_python_api_tutorial.html for how to use it

closed 2023-04-12 by JafarAbdi