moveit/moveit2 #1840: moveit_servo build failures on Rolling

nuclearsandwich, 2023-01-04:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build? binary binary build on build.ros2.org
* If binary, which release version? 2.6.0-1
* If source, which branch? N/A

### Links to failing builds

* https://build.ros2.org/view/Rbin_uJ64/job/Rbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/102/

### Backtrace or Console output

```
00:04:52.920 /usr/lib/ccache/c++ -DBACKWARD_HAS_DW=1 -DBOOST_ALL_NO_LIB -DBOOST_ATOMIC_DYN_LINK -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DDEFAULT_RMW_IMPLEMENTATION=rmw_fastrtps_cpp -DFMT_LOCALE -DFMT_SHARED -DRCUTILS_ENABLE_FAULT_INJECTION -Dmoveit_servo_lib_EXPORTS -I/tmp/binarydeb/ros-rolling-moveit-servo-2.6.0/include -isystem /opt/ros/rolling/include/control_msgs -isystem /opt/ros/rolling/include/geometry_msgs -isystem /opt/ros/rolling/include -isystem /opt/ros/rolling/include/moveit_msgs -isystem /opt/ros/rolling/include/pluginlib -isystem /opt/ros/rolling/include/rclcpp -isystem /opt/ros/rolling/include/rclcpp_components -isystem /opt/ros/rolling/include/sensor_msgs -isystem /opt/ros/rolling/include/std_msgs -isystem /opt/ros/rolling/include/std_srvs -isystem /opt/ros/rolling/include/tf2_eigen -isystem /opt/ros/rolling/include/trajectory_msgs -isystem /usr/include/eigen3 -isystem /usr/include/bullet -isystem /opt/ros/rolling/include/rcutils -isystem /opt/ros/rolling/include/rclcpp_action -isystem /opt/ros/rolling/include/ament_index_cpp -isystem /opt/ros/rolling/include/lifecycle_msgs -isystem /opt/ros/rolling/include/rclcpp_lifecycle -isystem /opt/ros/rolling/include/rcpputils -isystem /opt/ros/rolling/include/controller_manager_msgs -isystem /opt/ros/rolling/include/diagnostic_msgs -isystem /opt/ros/rolling/include/action_msgs -isystem /opt/ros/rolling/include/builtin_interfaces -isystem /opt/ros/rolling/include/rosidl_runtime_c -isystem /opt/ros/rolling/include/rosidl_typesupport_interface -isystem /opt/ros/rolling/include/unique_identifier_msgs -isystem /opt/ros/rolling/include/fastcdr -isystem /opt/ros/rolling/include/rosidl_runtime_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_cpp -isystem /opt/ros/rolling/include/rmw -isystem /opt/ros/rolling/include/rosidl_typesupport_fastrtps_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_c -isystem /opt/ros/rolling/include/rosidl_typesupport_introspection_cpp -isystem /opt/ros/rolling/include/libstatistics_collector -isystem /opt/ros/rolling/include/rcl -isystem /opt/ros/rolling/include/rcl_interfaces -isystem /opt/ros/rolling/include/rcl_logging_interface -isystem /opt/ros/rolling/include/rcl_yaml_param_parser -isystem /opt/ros/rolling/include/libyaml_vendor -isystem /opt/ros/rolling/include/tracetools -isystem /opt/ros/rolling/include/statistics_msgs -isystem /opt/ros/rolling/include/rosgraph_msgs -isystem /opt/ros/rolling/include/rosidl_typesupport_cpp -isystem /opt/ros/rolling/include/rosidl_typesupport_c -isystem /opt/ros/rolling/include/angles -isystem /opt/ros/rolling/include/shape_msgs -isystem /opt/ros/rolling/include/object_recognition_msgs -isystem /opt/ros/rolling/include/octomap_msgs -isystem /usr/include/libqhull_r -isystem /opt/ros/rolling/include/resource_retriever -isystem /opt/ros/rolling/include/visualization_msgs -isystem /opt/ros/rolling/include/urdf -isystem /opt/ros/rolling/include/urdf_parser_plugin -isystem /opt/ros/rolling/include/urdfdom_headers -isystem /opt/ros/rolling/include/urdfdom -isystem /opt/ros/rolling/include/class_loader -isystem /opt/ros/rolling/include/tf2 -isystem /opt/ros/rolling/include/tf2_ros -isystem /opt/ros/rolling/include/message_filters -isystem /opt/ros/rolling/include/rcl_action -isystem /opt/ros/rolling/include/tf2_msgs -isystem /opt/ros/rolling/include/tf2_geometry_msgs -isystem /opt/ros/rolling/include/kdl_parser -isystem /opt/ros/rolling/include/rsl -isystem /opt/ros/rolling/include/rcl_lifecycle -isystem /opt/ros/rolling/include/composition_interfaces -g -O2 -ffile-prefix-map=/tmp/binarydeb/ros-rolling-moveit-servo-2.6.0=. -flto=auto -ffat-lto-objects -flto=auto -ffat-lto-objects -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Wall -Wextra -Wwrite-strings -Wunreachable-code -Wpointer-arith -Wredundant-decls -Wcast-qual -Wno-unused-parameter -Wno-unused-function -Wno-maybe-uninitialized -std=c++17 -MD -MT CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o -MF CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o.d -o CMakeFiles/moveit_servo_lib.dir/src/servo_calcs.cpp.o -c /tmp/binarydeb/ros-rolling-moveit-servo-2.6.0/src/servo_calcs.cpp
00:04:52.922 /tmp/binarydeb/ros-rolling-moveit-servo-2.6.0/src/servo_calcs.cpp:45:10: fatal error: controller_manager/realtime.hpp: No such file or directory
00:04:53.550    45 | #include <controller_manager/realtime.hpp>
00:04:53.550       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
00:04:53.550 compilation terminated.
```

AndyZe, 2023-01-04:
Thanks. Should be fixed by this PR:  https://github.com/ros-planning/moveit2/pull/1791

I just rebased, let's see if it passes CI.

AndyZe, 2023-01-04:
OK, the MoveIt PR to fix this didn't pass. I think the root cause is that realtime-tools was released from the wrong branch. You can see that in rosdistro. Note the foxy-devel. So, the release doesn't have the new feature that's needed.

https://github.com/ros/rosdistro/blob/master/rolling/distribution.yaml

```
  realtime_tools:
    doc:
      type: git
      url: https://github.com/ros-controls/realtime_tools.git
      version: foxy-devel
    release:
      tags:
        release: release/rolling/{package}/{version}
      url: https://github.com/ros2-gbp/realtime_tools-release.git
      version: 2.4.0-1
```

AndyZe, 2023-01-05:
Should be fixed with this PR -- https://github.com/ros/rosdistro/pull/35761

Can you possibly release the debians again for that package?  🙏

nuclearsandwich, 2023-01-06:
> OK, the MoveIt PR to fix this didn't pass. I think the root cause is that realtime-tools was released from the wrong branch. You can see that in rosdistro. Note the foxy-devel. So, the release doesn't have the new feature that's needed.

Thanks for taking a look @AndyZe, the release version: 2.4.0 is the latest version of realtime_tools's `master` branch whereas the `foxy-devel` branch's latest release is 2.2.0 so I think the binary release is correct but the source entry was indeed out of date and the linked PR has been merged to fix that.

AndyZe, 2023-01-19:
The Humble build failure should be fixed when this goes through:  https://github.com/ros/rosdistro/commit/c17c002acf3968f5946a906e32d6066ccd6af358

henningkayser, 2023-01-26:
Humble seems to be back to normal: I'm closing this https://build.ros2.org/job/Hbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/

henningkayser, 2023-01-26:
ok, that was premature, rolling still is broken https://build.ros2.org/job/Hbin_uJ64__moveit_servo__ubuntu_jammy_amd64__binary/... Are we just waiting for a sync?

henningkayser, 2023-01-26:
@AndyZe with https://github.com/ros-planning/moveit2/pull/1791 we need a new release to get this fixed. How is the situation in humble and foxy, will we have to backport this or can we rely on stable ros2_control dependencies?

closed 2023-03-02 by AndyZe