moveit/moveit2 #1841: Octomap feature: keep "contiguous voxels" only

AndyZe, 2023-01-04:
A lot of the time, the pointcloud inputs that are converted to Octomap are noisy. There are stray voxels here and there, which end up getting converted to stray Octomap cubes.

What if there were an option to keep contiguous voxels only? I mean, only keep the octomap cubes which touch other octomap cubes. I think this would be great for most use cases like picking from a tabletop, because everything on the table and the table itself are contiguous.

Tagging @sjahr

ummy11, 2023-01-06:
These voxels could be a result of noisy measurement from the sensor.
You check to see if you have a pointcloud in the vicinity of the voxel in the octomap.
If that indeed is  the case, you could perhaps pre-filter the incoming pointcloud using something like a statistical outlier filter 
https://pcl.readthedocs.io/projects/tutorials/en/latest/statistical_outlier.html

pathakarvindk, 2023-01-06:
@AndyZe 
@schornakj 
@ummy11 
@rojkov 
@garaemon

Hey Guys,
I am new to ROS2 environment.
How to get octomap in rviz2 using a point cloud in moveit2 (ROS2 foxy)?

schornakj, 2023-01-06:
@pathakarvindk could you create a new, separate issue for your question?

AndyZe, 2023-01-06:
@pathakarvindk there is a tutorial for that. It might only be available as a ROS1 tutorial, though. I'm not sure it's been updated for ROS2.

I'll close this issue b/c everybody so far has suggested to pre-filter the pointcloud.

closed 2023-01-06 by AndyZe