moveit/moveit2 #1846: Octomap through point cloud in rviz2

pathakarvindk, 2023-01-06:
Hey Guys,
I am new to ROS2 environment.
How to get octomap in rviz2 using a point cloud in moveit2 (ROS2 foxy)?
Do we have any tutorial for the same?

Any help would be highly appreciated.

AndyZe, 2023-01-06:
The tutorial is not ported to ROS2 yet but it should be the same:

https://ros-planning.github.io/moveit_tutorials/doc/perception_pipeline/perception_pipeline_tutorial.html

grace-lovelace, 2023-01-26:
@pathakarvindk I would be happy to look at your solution if you wanna link your working branch.

I'd also be willing to help port the Perception Pipeline Tutorial to ROS2. I have real-world experience with all of the elements from the original tutorial, though my experience with ROS2 is limited. If I were to start writing a tutorial based on the original page, should I target Galactic or Humble? The [How To Contribute](https://github.com/ros-planning/moveit2_tutorials/blob/humble/doc/how_to_contribute/how_to_write_how_to_guides.rst) page lists Galactic, but I thought that recommendation might be out of date.

grace-lovelace, 2023-01-26:
I got started on this and came across something curious I thought I would loop back here. It turns out there is already a Perception Pipeline Tutorial for Humble on in [moveit2_tutorials](https://github.com/ros-planning/moveit2_tutorials/blob/humble/doc/examples/perception_pipeline/perception_pipeline_tutorial.rst). It looks like it was added just after these last comments in this thread from @pathakarvindk, though you have to dig a little to find them.

I would expect the Perception Pipeline Tutorial to be clearly listed beneath the Tutorials or Examples sections like it was in the ROS1 docs, but instead the Perception Pipeline Tutorial is listed as a step in the [Integration with a New Robot](https://moveit.picknik.ai/humble/doc/examples/examples.html#integration-with-a-new-robot) section (image below).

While accessibility might be a little wonky, I would say this tutorial does indeed exist and this issue can probably be closed. Do you have any thoughts on this, @AndyZe? I'm not a regular open-source contributor, so forgive me if the tagging is inappropriate.

[image]

peterdavidfagan, 2023-04-15:
Thanks for flagging this @grace-lovelace. I don't believe the tutorial has been ported yet, I am also not sure why this tutorial is present under the Integration with a new robot section. I need to clean up the Python API documentation, so my hands are somewhat full at the moment. With this being said, a demo I am currently working on leverages point cloud data for grasp pose prediction, adding an octomap to the planning scene is something I will be writing code for soon, if I have the time I'll open an pr to update this tutorial.

sjahr, 2023-06-15:
@pathakarvindk I hope your question is answered. Feel free to re-open if you need more help with this

closed 2023-06-15 by sjahr