moveit/moveit2 #1855: Any fixes for FAST DDS?

metanav, 2023-01-11:
ROS2 Humble
Moveit2 Humble
Ubuntu 22.04

I am using the Moveit2  "Pick and Place with MoveIt Task Constructor" tutorial which only works with Cyclone DDS.
Also, I am using a microROS node which seems not compatible with the Cyclone DDS and not working at all.
Which Moveit2 version works well with FAST DDS? Any future version has the fixes? I need to know these things before implementing microROS.

tylerjw, 2023-01-11:
Right now, it seems like FastDDS has issues we don't understand. Interestingly if you build fastDDS from source with some of the older versions of it, it works fine. The problem is with some of the newer versions of fastDDS. There is a chance the problem is the code in MoveIt and we are not using ROS correctly in some way, causing this incompatibility. We don't yet know why and are investigating this. If there was something we could change to make it work we'd be really happy to do so.

tylerjw, 2023-01-11:
Tagging @sea-bass because he looked into this most recently.

We are looking into trying to isolate the issues. Here are some links to issues on fastrtps's repo that we are related to this:

- https://github.com/ros2/rmw_fastrtps/issues/656
- https://github.com/ros2/rmw_fastrtps/issues/650
- https://github.com/ros2/rmw_fastrtps/issues/635

MiguelCompany, 2023-01-11:
@tylerjw I think all those issues might have been solved by eProsima/Fast-DDS#3194 and its corresponding backports (eProsima/Fast-DDS#3195) for Humble.

Could you check?

metanav, 2023-01-11:
@MiguelCompany, Do I need to build eProsima/Fast-DDS 2.6.x from source or it's available with the fresh ROS2 Humble binaries installation?

metanav, 2023-01-12:
In the https://github.com/ros2/ros2/blob/humble/ros2.repos, the 2.6.x is already mentioned. It means if I build ROS2 Humble from source it will be automatically applied. Right?
```
  eProsima/Fast-DDS:
    type: git
    url: https://github.com/eProsima/Fast-DDS.git
    version: 2.6.x
```

MiguelCompany, 2023-01-12:
@metanav Yes

metanav, 2023-01-15:
It seems working now. Thanks!

closed 2023-01-15 by metanav