moveit/moveit2 #1859: Interpolate trajectories at the desired control rate, so that ros2_control doesn't apply spline fitting

AndyZe, 2023-01-12:
ros2_control applies splines to any JTC trajectories it receives, but [there's an option to disable that](https://github.com/ros-controls/ros2_controllers/pull/374). I think we should disable it b/c spline fitting can cause overshooting and collisions.

But, then we need to make sure the trajectory message into ros2_control has waypoints spaced with the correct delta_t. This is what's missing right now. Currently, TOTG waypoint spacing defaults to 0.1 seconds.

destogl, 2023-01-12:
The only detail I am not certain how you want to solve here is trajectory replacement. i.e., trajectory changes during execution.

It could be very tricky to get this done properly in MoveIt since you are probably not having all data up to date. This means that you would have to everything in a controller step:

- get joint data
- generate trajectory
- do TOTG
- do Ruckig
- send to JTC

Maybe there are optimization possibilities like generating short trajectory, but this could end-up with online-planning all the time. Which is fine with me, but could cause a lot of issues on a not high-level HW.

AndyZe, 2023-01-12:
Closing in favor of the discussion here:  https://github.com/ros-planning/moveit2/discussions/1860

closed 2023-01-12 by AndyZe