moveit/moveit2 #1862: Remove last_plan_solution_ in PlanningComponent

sjahr, 2023-01-15:
Let us remove last_plan_solution_ from the PlanningComponent class once #1835 is merged. With the `execute()` method removed, the only actual usage of last_plan_solution_ within the class is removed. It would simplify this class and I think it is better to let the user of PlanningComponent handle the lifetime of results from `plan()` rather than having a member in the planning component who's state might be unknown.
What do you think @henningkayser @AndyZe ?

AndyZe, 2023-01-15:
I'd agree with that because it seems this is not used anywhere:

```
const planning_interface::MotionPlanResponse& PlanningComponent::getLastMotionPlanResponse()
{
  return last_plan_solution_;
}
```

henningkayser, 2023-01-19:
I like this idea a lot. Without a meaningful state the only real use case of the PlanningComponent is building of motion plan requests, which should in turn be addressed by #1455. I think that way the API would become much easier to use and to extend.

sjahr, 2023-01-19:
I'll create a PR. Small annoyance: This will definitely break the API since
```
[[deprecated]]
bool PlanningComponent::execute(bool blocking);
```
wont work anymore

sjahr, 2023-03-20:
Closed with #1879

closed 2023-03-20 by sjahr