moveit/moveit2 #1863: Control robot in ROS1 Noetic using MoveIt2 Foxy

caiobarrosv, 2023-01-16:
Hi,

I'd like to control a robot in ROS1 Noetic by using MoveIt2 Foxy as shown in this [video](https://www.youtube.com/watch?v=Q-0SFr8uO64).

I am using using a ros bridge to bridge `topics` and also `actions`.
I can see the topics `/joint_states`, `/clock`, `/tf_static`  and also the action `/undefined_ur10/arm_controller/command` in ROS2.

In addition, the robot is shown in RVIZ2 (Foxy) with the goal gizmo but I can't still control the robot in ROS1 (Noetic).

[image]

What are the configuration files I need to tweak in order to complete this task?
Can someone give me some direction?

AndyZe, 2023-01-16:
I thought the bridge didn't support actions? @MikeWrock is probably the perfect person to answer you question, if anybody can

MikeWrock, 2023-01-16:
@AndyZe is correct in his statement, the bridge doesn't _officially_ support actions however there are a number of workarounds as well as some significant efforts to [add action bridging to the repo](https://github.com/ros2/ros1_bridge/pull/256) (along with a discourse discussion [here](https://discourse.ros.org/t/introducing-action-bridge/9103)).

As you may have discovered, you can still bridge the `your_action/command` topic and achieve similar functionality to a bridged action but if you're looking for the full functionality of the action server you may have to make use of the PR I linked.

Feel free to share an example and I'd be happy to try it out, sometimes it can be as simple as trying out a different DDS implementation

caiobarrosv, 2023-01-16:
Thank you for answering @AndyZe  and @MikeWrock.

Actually I am using a non-official version of a action bridge. I can see the `/arm_controller/follow_joint_trajectory_action` in both ROS1 and ROS2.

```bash
$ ros2 action list
/arm_controller/follow_joint_trajectory_action
```

```bash
ros2 action info /arm_controller/follow_joint_trajectory_action 
Action: /arm_controller/follow_joint_trajectory_action
Action clients: 0
Action servers: 1
    /ros_bridge_action
```

I teste the communication between the action client in ROS2 and the action server in ROS1 and it worked. 
I sent some trajectory points and the robot moved.

My problem now is that MoveIt2 is not working eventhough I set the correct parameters for the controllers.

I get an error when trying to send a goal using MoveIt2:
```
[move_group-4] [ERROR] [1673911003.704454720] [moveit_ros.trajectory_execution_manager]: Unable to identify any set of controllers that can actuate the specified joints: 
[ elbow_joint shoulder_lift_joint shoulder_pan_joint wrist_1_joint wrist_2_joint wrist_3_joint ]
```

I tried to solve the issue as proposed [here](https://github.com/ros-planning/moveit2/issues/1394) but didn't work.

I think this is last step to get this working since the action client in ROS2 is already sending joint commands and moving the robot in ROS1.

caiobarrosv, 2023-01-17:
I've got this working.

I added the following controller to the controllers:
```yaml
controller_manager:
  ros__parameters:
    update_rate: 600  # Hz

    arm_controller:
          type: joint_trajectory_controller/JointTrajectoryController

arm_controller:
  ros__parameters:
    joints:
        - shoulder_pan_joint
        - shoulder_lift_joint
        - elbow_joint
        - wrist_1_joint
        - wrist_2_joint
        - wrist_3_joint
        - finger_joint
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
    state_publish_rate: 100.0
    action_monitor_rate: 20.0
    allow_partial_joints_goal: false
    constraints:
      stopped_velocity_tolerance: 0.0
      goal_time: 0.0
```

and also the controller with the proper action_ns:

```yaml
controller_names:
  - arm_controller
  - joint_trajectory_controller

arm_controller:
  action_ns: follow_joint_trajectory_action
  type: FollowJointTrajectory
  default: true
  joints:
    - shoulder_pan_joint
    - shoulder_lift_joint
    - elbow_joint
    - wrist_1_joint
    - wrist_2_joint
    - wrist_3_joint
    - finger_joint
```

Is action is being bridged between ROS1 and ROS2.
After I load all the controllers in the launch file, now MoveIt2 is able to send commands to the ROS1 Robot Simulator.

[image]

AndyZe, 2023-01-17:
congratulations! It looks nice.

closed 2023-01-17 by AndyZe