moveit/moveit2 #1869: Unable to build MoveIt2 from source on rolling

Wolf14, 2023-01-18:
### Description
So I was following the instructions at https://moveit.ros.org/install-moveit2/source/ to build from source. 
Colcon fails to build MoveIt2 from source. Initially the build failed due to missing package dependencies such as object_recognition_msgs, angles and ruckig. After adding these manually, the build failed while building moveit_core.

Background on me: I'm new to open source and build systems in general. I tried finding the solution to this online for a long time but couldn't.

### Your environment
* ROS Distro: Rolling
* OS Version: e.g. Ubuntu 22.04
* Source build
* Main branch
* Cyclone DDS.

### Steps to reproduce
* Build ROS rolling from source in non standard directory(using instructions from https://docs.ros.org/en/rolling/Installation/Alternatives/Ubuntu-Development-Setup.html).
* Build MoveIt2 from source.
* git clone and add the repositories it shows is missing when the build fails like object_recognition_msgs, angles and ruckig (dont * remember all of them which I had added like this).

### Expected behaviour
Building without any errors

### Actual behaviour
CMake generate step fails saying build files cannot be regenerated correctly.

### Backtrace or Console output
https://gist.github.com/Wolf14/7ac2f1d425f9a09248b6c55125423fd0

cottsay, 2023-01-19:
This describes the same issue as #1858.

vatanaksoytezer, 2023-01-19:
Friendly tagging @tylerjw and @ChrisThrasher for some visibility to this issue.

henningkayser, 2023-01-19:
Hi @Wolf14, could you try if #1872 fixes your issue and close it if it does?

Wolf14, 2023-01-21:
Hey thanks! #1872 worked I was able to build moveit_core.

closed 2023-01-21 by Wolf14