moveit/moveit2 #1870: Build error caused by missing some OpenMP flags

danielemorra98, 2023-01-18:
### Description
The `colcon build` command found [here](https://moveit.ros.org/install-moveit2/source/) fails because CMake cannot find OpenMP_C_FLAGS, OpenMP_C_LIB_NAMES, OpenMP_CXX_FLAGS and OpenMP_CXX_LIB_NAMES flags. 
After some research, I found that the issue was prompted by different version of openMP (in my case I had libomp-dev version 10) and the compiler (clang-12 and clang++-12). I got the same errors when I switched to GNU 9.4 compiler. 

### Your environment
* ROS Distro: Galactc
* OS Version: e.g. Ubuntu 20.04
* Source build
* Compiler: clang-12

### Solution (for Clang)
As suggested [here](https://stackoverflow.com/questions/69797680/cmakes-find-packageopenmp-cannot-find-openmp-when-using-clang-12) align the version of both compiler and library

henningkayser, 2023-01-19:
Hi @danielemorra98, are the install instructions recommending anywhere to use `clang-12` with Focal? For now, we mostly stuck with the distro defaults that would also be resolved by rosdep accordingly: `clang` -> `clang-10` and `libomp-dev` -> `libomp-10-dev`. In Jammy, both resolve to version 12. Using version 12 in Focal is of course possible, but it's up to the user to ensure all dependencies are aligned, I don't see a straight-forward way to do this (we could set cmake guards, but that would still not solve the issue that there is no way to tell rosdep to install the right package version). Note, that the galactic docker still runs on 10 (https://github.com/ros-planning/moveit2/blob/galactic/.docker/ci/Dockerfile#L28) and that pre-commit also enforces formatting with `clang-format-10` (https://github.com/ros-planning/moveit2/blob/galactic/.pre-commit-config.yaml#L45).

Since Galactic has also reached EOL, I would recommend switching to Humble or even Rolling for getting all the latest versions supported.

danielemorra98, 2023-01-24:
Thanks for your answer. I agree with you that it is up to the developer resolve the issue with deps. Anyway I would cite this on the [Build page](https://moveit.ros.org/install-moveit2/source/). There is no mention to any versions of Clang.

JafarAbdi, 2023-04-12:
Closing since galactic reached its EOL date

closed 2023-04-12 by JafarAbdi