moveit/moveit2 #1878: RViz2 SegFault with "Use Cartesian Path" on the Tutorial

yashi, 2023-01-20:
### Description

RViz2 dies with Segmentation Fault when I check "Use Cartesian Path" option in the [tutorial](https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#plan-cartesian-motions).  

It seems you get `nullptr` as `reference_state` in `RobotTrajectory::setRobotTrajectoryMsg()`.  And touching it in `RobotState::RobotState()`.

I'm full building the stack on an unsupported Debian Sid.  So it might be just me.

### Your environment
* ROS Distro: Humble
* OS Version: Debian Sid
* Source: Source build from scratch
* If source, which branch? a4f7af6ce6466e4b9e989c7709feca2f44a11cc6
* Which RMW: Cyclone DDS

### Steps to reproduce

I'm testing https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html#plan-cartesian-motions. Which means

1. `ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true`
3. Select "Use Cartesian Path"
4. Hit "Plan" button

### Expected behaviour
Not Segfaults

### Actual behaviour
RViz2 dies with `[ERROR] [rviz2-1]: process has died [pid 3680372, exit code -11`

### Backtrace or Console output
https://gist.github.com/yashi/8ddd3efa283202eecf4ad4d31baf89a4

```
(gdb) f 10
#10 0x00007f6e30058db3 in robot_trajectory::RobotTrajectory::setRobotTrajectoryMsg (this=0x7f6de11fd100, reference_state=..., trajectory=...)
    at /home/yashi/work/yoshida/moveit-ws/src/moveit2/moveit_core/robot_trajectory/src/robot_trajectory.cpp:421
421	    auto st = std::make_shared<moveit::core::RobotState>(copy);
(gdb) p reference_state
$2 = (const moveit::core::RobotState &) <error reading variable: Cannot access memory at address 0x0>
(gdb) 
```

yashi, 2023-01-20:
BTW, it works fine if you uncheck "Use Cartesian Path".

yashi, 2023-01-22:
I just tried on Ubuntu 22.04 and it crashes as well.

Abishalini, 2023-08-22:
Hi @yashi ,
I was not able to reproduce this issue. I'm using the latest rolling to test.
https://github.com/ros-planning/moveit2/assets/17363793/11c53e37-ef08-4036-8112-146b3d503c70

Can you let me know if you had other settings differences to reproduce this?

yashi, 2023-08-23:
@Abishalini Thank you for your reply. I'll find time to test it again wtih rolling.

yashi, 2023-09-12:
@Abishalini, I've rebuilt entire moveit stack with the current tip for Humble and I don't see the segfault.  I'm closing this.

closed 2023-09-12 by yashi

dan9thsense, 2024-04-12:
Exact same problem with Iron.