moveit/moveit2 #1885: Servo: Returns to last sent position when unpaused

HX2003, 2023-01-24:
If joint positions move by some other method when servo is paused or commands are stale, upon receiving new commands that are zero it returns to last sent position. Hence, there is a sudden movement; I don't think that is supposed to be the intended behaviour?
https://github.com/ros-planning/moveit2/blob/e2e0003c0e170a05747c699da05258382a338edf/moveit_ros/moveit_servo/src/servo_calcs.cpp#L452-L453

AndyZe, 2023-01-24:
Thanks for tracking down one of the lines involved. I'll file a PR to fix this soon. Can you test it for me?

henningkayser, 2023-01-26:
Is this issue related to https://github.com/ros-planning/moveit2/issues/1036?

AndyZe, 2023-01-27:
I'm starting to look at this. Servo has these flags, `latest_twist_cmd_is_nonzero_` and `latest_joint_cmd_is_nonzero_`. The original thinking was, Servo should skip publishing commands if the user is requesting zero velocity. This was done to reduce spammy messages to ros_control.

I'm now thinking, those flags should just be dropped. No harm done if Servo publishes zero-velocity commands always.

AndyZe, 2023-01-27:
@HX2003 can you please test #1904?

HX2003, 2023-01-27:
@AndyZe I can confirm that the proposed fix works as intended. Additionally, can you expose the paused/unpaused status as a topic, I have multiple nodes that need to know whether servo is active.

AndyZe, 2023-01-27:
Isn't there a topic for this already, something like `/servo_server/status`?

Thanks for testing it so quickly!

HX2003, 2023-01-27:
There is a topic similar to it, but it does not reflect anything about pause status.
https://github.com/ros-planning/moveit2/blob/2fe4b30266fff91580234cbe0f05c6d1e1cefe8e/moveit_ros/moveit_servo/include/moveit_servo/status_codes.h

AndyZe, 2023-01-27:
OK. Well, I probably won't have time to make that PR in the next couple weeks. If you can make, I'll review, though.

Feel free to open a new issue requesting the feature, too.

HX2003, 2023-01-27:
Would it just involve publishing the paused status here as a separate topic? In that case I should be able to make a PR. And I should close this issue soon.
https://github.com/ros-planning/moveit2/blob/2fe4b30266fff91580234cbe0f05c6d1e1cefe8e/moveit_ros/moveit_servo/src/servo_calcs.cpp#L1232-L1235

AndyZe, 2023-01-27:
I think it would be nice if you could add a Paused status code to this list -- https://github.com/ros-planning/moveit2/blob/2fe4b30266fff91580234cbe0f05c6d1e1cefe8e/moveit_ros/moveit_servo/include/moveit_servo/status_codes.h

And publish to the same topic, `servo_server/status`.

Basically, I don't see why a new topic is necessary ?

HX2003, 2023-01-27:
Alright, I'll see what I can so.

closed 2023-02-03 by HX2003