moveit/moveit2 #1895: Flaky push/pull success when pushing to Docker registry

henningkayser, 2023-01-26:
We have been seeing weird issues when pulling and pushing docker images in CI jobs (and also locally, at least for me).

Examples
* Push: https://github.com/ros-planning/moveit2/actions/runs/4014707063/jobs/6895553038#step:5:6364
```
Error: buildx failed with: ERROR: failed to solve: failed to push ghcr.io/ros-planning/moveit2:rolling-ci: failed to copy: io: read/write on closed pipe
```
* Pull: https://github.com/ros-planning/moveit2/actions/runs/4014707057/jobs/6895552683#step:12:36
```  $ docker pull ghcr.io/ros-planning/moveit2:rolling-ci
  rolling-ci: Pulling from ros-planning/moveit2
  manifest unknown
  'docker pull ghcr.io/ros-planning/moveit2:rolling-ci' returned with 1
  'pull_docker_image' returned with code '1' after 0 min 0 sec
```

I'm still trying to make sense of this. Re-triggering CI already succeeded the jobs for a couple times, but now it seems to fail consistently, at least for most build jobs.

vatanaksoytezer, 2023-01-26:
@henningkayser should we fall back to docker.io until this is resolved?

vatanaksoytezer, 2023-01-26:
https://quay.io/ also offers free storage for open-source repositories and has been fairly stable in my experience.

henningkayser, 2023-04-06:
Possibly related issue https://github.com/docker/build-push-action/issues/851