moveit/moveit2 #1907: Exception while loading move_group capability

metanav, 2023-01-29:
### Description
I am trying to run moveit2_tutorials pick and place with task constructor demo. It was working 1 month ago but after rebuilding latest moveit2 from the source it is not working. 

### Your environment
* ROS Distro: Humble
* OS Version: MacOS Ventura 13.1
* Source build Humble branch
* Which RMW Fast DDS or Cyclone DDS)? Fast DDS

### Steps to reproduce
```
$ ros2 launch moveit2_tutorials mtc_demo.launch.py

```
### Expected behaviour
It should load all move_group capabilities.

### Actual behaviour
I am getting following errors in the console log.

```
[move_group-4] [INFO] [1674950518.255124082] [move_group.move_group]: MoveGroup debug mode is ON
[move_group-4] [ERROR] [1674950519.556807809] [move_group.move_group]: Exception while loading move_group capability 'move_group/ApplyPlanningSceneService': MultiLibraryClassLoader: Could not create object of class type move_group::ApplyPlanningSceneService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.560767024] [move_group.move_group]: Exception while loading move_group capability 'move_group/ClearOctomapService': MultiLibraryClassLoader: Could not create object of class type move_group::ClearOctomapService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.893477601] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupCartesianPathService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupCartesianPathService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.898292080] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupExecuteTrajectoryAction': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupExecuteTrajectoryAction as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.902719780] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupGetPlanningSceneService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupGetPlanningSceneService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.906225898] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupKinematicsService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupKinematicsService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.909389472] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupMoveAction': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupMoveAction as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.912368214] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupPlanService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupPlanService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.915718827] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupQueryPlannersService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupQueryPlannersService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [ERROR] [1674950519.920135486] [move_group.move_group]: Exception while loading move_group capability 'move_group/MoveGroupStateValidationService': MultiLibraryClassLoader: Could not create object of class type move_group::MoveGroupStateValidationService as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[move_group-4] [INFO] [1674950519.920173976] [move_group.move_group]: 
[move_group-4] 
[move_group-4] ********************************************************
[move_group-4] * MoveGroup using: 
[move_group-4] *     - ExecuteTaskSolution
[move_group-4] ********************************************************
[move_group-4] 
[move_group-4] [INFO] [1674950519.920207499] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[move_group-4] [INFO] [1674950519.920226130] [moveit_move_group_capabilities_base.move_group_context]: MoveGroup context initialization complete
[move_group-4] Loading 'move_group/ApplyPlanningSceneService'...
[move_group-4] Loading 'move_group/ClearOctomapService'...
[move_group-4] Loading 'move_group/ExecuteTaskSolutionCapability'...
[move_group-4] Loading 'move_group/MoveGroupCartesianPathService'...
[move_group-4] Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
[move_group-4] Loading 'move_group/MoveGroupGetPlanningSceneService'...
[move_group-4] Loading 'move_group/MoveGroupKinematicsService'...
[move_group-4] Loading 'move_group/MoveGroupMoveAction'...
[move_group-4] Loading 'move_group/MoveGroupPlanService'...
[move_group-4] Loading 'move_group/MoveGroupQueryPlannersService'...
[move_group-4] Loading 'move_group/MoveGroupStateValidationService'...
[move_group-4] 
[move_group-4] You can start planning now!

```

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

metanav, 2023-02-05:
I removed ROS2 humble and MoveIt2 and did a fresh installation but still have same issue. Any hints where should I look to fix these errors?

JafarAbdi, 2023-02-05:
Have you tried switching to cyclonddds? https://github.com/ros-planning/moveit2/issues/1874

metanav, 2023-02-06:
> Have you tried switching to cyclonddds? #1874

I have already tried with `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp` but it did not work.

metanav, 2023-02-21:
Any updates on this?

JafarAbdi, 2023-04-12:
@metanav Sorry I didn't have time to look into this. Were you able to fix the issue?

okvik, 2023-05-19:
@JafarAbdi I'm encountering the same issue. Took me a while to find this report. Initially I thought the issue might be caused by something in my ROS package management setup (using Nix, i.e. `nix-ros-overlay`), but apparently not. Here's my issue in the packages repo: https://github.com/lopsided98/nix-ros-overlay/issues/264

An interesting thing I encountered is that listing capabilities through provided utility works just fine, while move_group itself doesn't find the same plugins.
 
```
$ ros2 run moveit_ros_move_group list_move_group_capabilities

Available capabilities:
move_group/ApplyPlanningSceneService
move_group/ClearOctomapService
move_group/MoveGroupCartesianPathService
move_group/MoveGroupExecuteService
move_group/MoveGroupExecuteTrajectoryAction
move_group/MoveGroupGetPlanningSceneService
move_group/MoveGroupKinematicsService
move_group/MoveGroupMoveAction
move_group/MoveGroupPlanService
move_group/MoveGroupQueryPlannersService
move_group/MoveGroupStateValidationService
move_group/TfPublisher
pilz_industrial_motion_planner/MoveGroupSequenceAction
pilz_industrial_motion_planner/MoveGroupSequenceService
```

I'm gonna try using rolling for the moment. I'll also try changing the RMW, as mentioned above.

okvik, 2023-05-19:
Update: neither switching to rolling release nor changing RMW helped the case. I'll dive deeper next week.

metanav, 2023-07-13:
@okvik any updates?

okvik, 2023-07-13:
@metanav Sorry, I haven't found further clues regarding this problem.

FWIW MoveIt2 works just fine for me in a standard documented (Ubuntu) setup, I've only had this issue with the Nix packaging of it.

sjahr, 2023-08-09:
@metanav I hope your issue is resolved by now. If not feel free to re-open this issue if it is not the case.

closed 2023-08-09 by sjahr

metanav, 2023-11-18:
It's not resolved yet. Recently I have done a fresh installation on a new macOS Sonoma 14.1.1 but it shows same errors.

rcruzoliver, 2024-01-15:
Hello everyone, 

I had this issue and found the reason behind. The main branch of [moveIt tutorial](https://github.com/ros-planning/moveit2_tutorials) is not correct for "humble". 

When following this [tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) you should clone the "humble" branch. So

```bash
git clone -b humble https://github.com/ros-planning/moveit2_tutorials
```
and then:
```bash
vcs import < moveit2_tutorials/moveit2_tutorials.repos
sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
```
Be sure you start with a clean workspace and do not try to reuse the old one. 

Regards, 
Raul

usmanbiu, 2024-06-25:
> Hello everyone,
> 
> I had this issue and found the reason behind. The main branch of [moveIt tutorial](https://github.com/ros-planning/moveit2_tutorials) is not correct for "humble".
> 
> When following this [tutorial](https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html) you should clone the "humble" branch. So
> 
> ```shell
> git clone -b humble https://github.com/ros-planning/moveit2_tutorials
> ```
> 
> and then:
> 
> ```shell
> vcs import < moveit2_tutorials/moveit2_tutorials.repos
> sudo apt update && rosdep install -r --from-paths . --ignore-src --rosdistro $ROS_DISTRO -y
> ```
> 
> Be sure you start with a clean workspace and do not try to reuse the old one.
> 
> Regards, Raul

I had the same issue and your suggestion to clone the humble branch worked. The only difference is that in my case, I had cloned the moveit2 repo in an already existing workspace with many packages. I only had to delete the install and build folders in my workspace, I built the workspace again before closing moveit2 to a folder I created (in my workspace). Everything worked fine after that. 

PS. I followed the instructions in the moveit2 documentation for my installation but I cloned the humble branch of moveit2

rcruzoliver, 2024-06-26:
> I had the same issue and your suggestion to clone the humble branch worked. The only difference is that in my case, I had cloned the moveit2 repo in an already existing workspace with many packages. [...]

That's great. What I meant as "clean workspace" is exactly what you did: delete the install and build directories...colcon isnt otherwise fresh building the package with the new code.