moveit/moveit2 #1920: Replace unhelpful cmake variable PROJECT_NAME with actual package name

tylerjw, 2023-02-03:
[image]

`${PROJECT_NAME}` is a variable that is found in many of our cmake files. [here is one example](https://github.com/ros-planning/moveit2/blob/2105b91f89ea213a0fc3fc6b3735d296b5d57c3f/moveit_kinematics/CMakeLists.txt#L68). The project name in the lined to example is found at the top of the file `project(moveit_kinematics)`. It would make this file more readable if all instances of `${PROJECT_NAME}` was replaced with `moveit_kinematics`.

1. [ ] Find `${PROJECT_NAME}` in cmake files
2. [ ] Replace with the project name.
3. [ ] Be Happy!

rudra2112, 2023-02-05:
Hi, I am new to open source. I would like to work on this issue, can you assign me this issue, and help me a little more where should I start working from.

tylerjw, 2023-02-05:
You can't do it. Let me know what questions you have.

rudra2112, 2023-02-05:
I have fixed a lot of instances in PR[#1929](https://github.com/ros-planning/moveit2/pull/1929) can you check it and let me know, how can I improve.

AdarshSingh-14, 2023-03-06:
Is the issue fixed? if not and no one is working on it I would like to help with this.

henningkayser, 2023-03-10:
It looks like we now have two open PRs more or less applying the same fixes #1978, #1929

PR #2020 (2023-04-20): Replace Variable PROJECT_NAME in CMakeLists.txt with the actual name
### Description

Replaced PROJECT_NAME variable with the contents of the project(*) in CMakeLists.txt of various packages.
Fixes #1920.

### Checklist
- [x] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) notes
- [ ] Cre [... cut]
closed 2023-04-20 by tylerjw