moveit/moveit2 #1921: Create first-timer issues for CMake cleanup of single use variables

tylerjw, 2023-02-03:
There are places in our cmake files you can find lists that are defined in one place and used in only one other place. An example is `THIS_PACKAGE_LIBRARIES` in [moveit_kinematics cmake file](https://github.com/ros-planning/moveit2/blob/main/moveit_kinematics/CMakeLists.txt#L67). Moving the values in the list to the one place the variable is used is a nice cleanup as it makes the cmake file more declarative, shorter, and easier to read.

To create opportunities for new contributors and be a good mentor:

- [ ] Find instances of this in our CMake files
- [ ] Create `good-first-issue` issue making reference to this issue. Make sure to add that label.
- [ ] Review PRs that people submit to do these cleanups
- [ ] Enjoy cleaning up MoveIt and helping new contributors!

Tejal-19, 2023-03-11:
Hey, would like to take up this issue. Can you please guide me as to how to proceed further.

aaryanmurgunde, 2024-04-23:
@tylerjw Is there a perticular reason for this cleanup? Because as per my understanding this way of setting the lists, is just a more readable code and helps to define in the start of the CMake file, Additionally making them a little more standardised.

tylerjw, 2024-04-24:
This can be a style choice and is somewhat subjective. I'm happily no longer writing CMake and don't care what you do here.