moveit/moveit2 #1922: Simplify good-first-issue template

tylerjw, 2023-02-03:
The [good first issue template](https://github.com/ros-planning/moveit2/blob/main/.github/ISSUE_TEMPLATE/first_timers_only.md) is huge making it a pain to use. One of the primary motivations for mentoring new contributors is to scale our efforts. Having this huge complex form to fill out is not helping us.

One of the reasons the issue template is really long is it includes explaining things like how to submit a pull request. That is information that is useful for all contributors and should be documented in a standard location. The standard location in GitHub is the `CONTRIBUTING.md` file. Ours links to our website, which annoyingly makes it more difficult to improve as one has to learn to contribute to the website to add useful instructions for new contributors.

Tasks:
- [ ] Write a much much shorter issue template
- [ ] Create [Discussion](https://github.com/ros-planning/moveit2/discussions) about migrating contribution instructions from the website to CONTRIBUTING.md
- [ ] Be happy!

I'm up for mentoring this issue. Please ask any questions you have.

Kzernobog, 2023-02-20:
Hey @tylerjw. I was wondering if anyone is onto this? If not, shall I give it a go?

solosangeeth, 2024-03-24:
Hello @tylerjw.  I would like to claim this issue. I am new to open source, and It would be great if I could get some guidance please. I am currently going through the guidance at the moment.

closed 2026-09-22 by nbbrooks