moveit/moveit2 #1923: Specify LANGUAGE CXX in cmake where appropriate

tylerjw, 2023-02-03:
Specifying that a package is a C++ package to cmake is helpful in speeding up build times as it lets cmake know that it only needs to verify you have a working C++ compiler before building the package. If you don't specify this it checks for all sorts of other things your build probably doesn't need. [here is an example of how you specify that a package is a C++ package to CMake](https://github.com/ros-controls/ros2_control/blob/c8b843a9d4ec11bc1c0efe4d3339bca2b02fd586/controller_interface/CMakeLists.txt#L2)

Tasks:
- [ ] Find CMake packages in MoveIt that are C++ packages
- [ ] Add the config to the CMake
- [ ] Be happy!

I'm up for mentoring this issue, please ask any questions you have.

sameergupta4873, 2023-02-05:
@tylerjw can I take this issue ?

tylerjw, 2023-02-05:
Yes, please do. Let me know if you have any questions.

tylerjw, 2023-02-07:
@sameergupta4873 fixed this issue, thank you!

closed 2023-02-07 by tylerjw

sameergupta4873, 2023-02-07:
> @sameergupta4873 fixed this issue, thank you!

It's my pleasure, contributing in here.