moveit/moveit2 #1924: Create custom robot config for testing to eliminate circular dependency on moveit_resources

tylerjw, 2023-02-03:
## The problem

We use robot configs from [moveit_resources](https://github.com/ros-planning/moveit_resources/tree/ros2) in our [testing](https://github.com/ros-planning/moveit2/blob/2105b91f89ea213a0fc3fc6b3735d296b5d57c3f/moveit_core/package.xml#L65). This may seem fine except if a user tries to launch moveit using the launch files in moveit_resources they will fail because the dependency on moveit is [commented out](https://github.com/ros-planning/moveit_resources/blob/edd4d059f41d421c16f7006ce14f840dc23d8643/panda_moveit_config/package.xml#L25). This was done so we could release these two packages into the build-farm with moveit depending on moveit_resources.

Here is an example of an error users get if they try to launch the [demo.launch.py](https://github.com/ros-planning/moveit_resources/blob/humble/panda_moveit_config/launch/demo.launch.py#L9):

```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): Caught exception when trying to load file of format [py]: No module named 'moveit_configs_utils'
```

One way this could be fixed is to fold the moveit_resources package into the moveit repo. This is undesirable because the panda robot is a vendor and we don't want any vendor code in the main moveit repo.

## The solution

@pac48 created this really awesome robot for testing ros2_control:
https://github.com/pac48/ros2_control_demos/tree/full-example-tutorial/r6bot

It would be really cool if we had something similar inside moveit itself. This would allow us to migrate our tests and tutorials to work off this custom robot. This custom robot could be updated to show the best way to integrate a custom robot and would be something we'd control completely for testing.

## Tasks

- [ ] Create a custom robot in moveit2 repo (migrating in the one from @pac48 is an option)
- [ ] Create issues for migrating the tests to use this new robot

MarqRazz, 2023-02-03:
My problem with choosing a "fake" robot is that it is not very likely to be included in other simulators like the UR5/Franka are in Gazebo/Ignition/Isaac/MuJoCo... and you can't run any tests on hardware.

I ran into this issue was because I was trying to run the [moveit_resources](https://github.com/ros-planning/moveit_resources/pull/158) version of the Panda in Isaac and don't want to have to build MoveIt from source for it to work.

@tylerjw also pointed out that [moveit2_tutorials](https://github.com/ros-planning/moveit2_tutorials) suffers from this same issue and had to copy all of the launch files for the Panda robot to make them work.

AndyZe, 2023-03-23:
Realistically, we should use a UR robot, if there's only going to be one robot. They were one of the first with a ROS2 package and they are the most common