moveit/moveit2 #1925: Plugin implementations should be moved out of moveit_core

tylerjw, 2023-02-03:
It would greatly simplify moveit_core if it was built as a single library instead of many small libraries linked together. One of the problems with this is that several plugins (not just plugin interfaces) are part of the moveit_core package. This got in the way of refactoring the build scripts for moveit_core into a single library, see #1801

Here is a directory for packages that contain moveit_plugins: https://github.com/ros-planning/moveit2/tree/main/moveit_plugins

Tasks:
- [ ] Migrate `collision_detection::CollisionDetectorBtPluginLoader` from moveit_core into moveit_plugins.
- [ ] Migrate `collision_detection::CollisionDetectorFCLPluginLoader` from moveit_core into moveit_plugins.
- [ ] Migrate `online_signal_smoothing::ButterworthFilterPlugin` from moveit_core into moveit_plugins.
- [ ] Update moveit_plugins [package.xml](https://github.com/ros-planning/moveit2/blob/main/moveit_plugins/moveit_plugins/package.xml) to depend on these new plugins.
- [ ] Search for any tests that previously depended on moveit_core for these plugins and update them to depend on their new packages.
- [ ] Be happy!

I'd be happy to mentor doing this project.

sameergupta4873, 2023-02-12:
@tylerjw can i take this issue ?

AndyZe, 2023-02-12:
Sorry, I think @ChrisThrasher is already working on it   :/

ChrisThrasher, 2023-02-12:
I am. It's turning out to be a really complicated problem given how parts of moveit_core depend on parts of these plugins.

ChrisThrasher, 2023-03-30:
I currently have no plans on coming back to this problem. My first attempt was fraught with issues due to moveit_core's complicated internal dependency graph. So long as major parts of moveit_core depend on aspects of these plugins then this refactor will be tricky.

tylerjw, 2023-03-30:
I'm becoming annoyed with this stale bot thing because these issues are less relevant to the project just because we don't have any new status on them.