moveit/moveit2 #1928: Fix overriding binary install

tylerjw, 2023-02-03:
Overriding is when you build a package from a source that is also installed as a binary. Currently, this does not work in MoveIt because of the way header files are found. The problem is that the binary header is found before the source header when building and if the source one is newer and has changes the compile will break in surprising ways.

#1901 is a good example of what needs to be done everywhere in moveit. 

Here are the colcon docs on overriding: https://colcon.readthedocs.io/en/released/user/overriding-packages.html#how-to-make-it-easier-for-your-users-to-override

Tasks:
- [ ] Search for cmake files that are missing this package directory
- [ ] Make PR fixing the issue

bailaC, 2023-05-30:
@tylerjw I think we can close this issue as it is already merged.

sjahr, 2023-12-11:
@mariobo8 Yes, go for it.

advait-zx, 2024-09-21:
is this issue still open?

mikeferguson, 2024-12-17:
Duplicate of https://github.com/moveit/moveit2/issues/3134 which has more recent documentation to work from

closed 2024-12-17 by mikeferguson