moveit/moveit2 #1930: MoveIt Python Library: Replace Traditional For Loops with Range-based For Loops

peterdavidfagan, 2023-02-06:
## Background

The MoveIt python binding code contains traditional for loops that are less readable than range-based for loops. This issue is used to track the removal of range-based for loops in the python library binding code.  

## Instructions
Replace traditional for loops with range-based for loops where possible. 

### 🤔 What you will need to know.

For details on range-based for loops you can consult: https://en.cppreference.com/w/cpp/language/range-for.

Sujeendra, 2023-02-11:
Hi, I have created a pull request for this issue. Can you please review and merge my changes?

peterdavidfagan, 2023-02-11:
Awesome, thanks @Sujeendra I will review this over the weekend along with @JafarAbdi's pr.

peterdavidfagan, 2023-02-13:
Thanks @Sujeendra I have cherry picked your changes and assigned you as the commit author on the pull request.

peterdavidfagan, 2023-02-13:
Reopening due to build failure upon testing: https://github.com/peterdavidfagan/moveit2/pull/12

Sujeendra, 2023-02-18:
Hi peter, Is there any documentation for building the code in local environment? this will help me build my code before check-in my changes.

sshaizkhan, 2023-02-21:
@peterdavidfagan I have created a PR for the same and the code can build successfully through `moveit/moveit2:humble-ci-testing`  docker image. Please check the description for more info. Thanks

peterdavidfagan, 2023-02-21:
Hi @Sujeendra,

Apologies for the delay in response. The following documentation outlines the process for performing a source build of MoveIt. Using docker as @sshaizkhan has is also a good choice for building MoveIt in an environment separate to the host environment. 

Thanks @sshaizkhan I will test this shortly also to confirm.

peterdavidfagan, 2023-02-23:
Thanks @sshaizkhan for resolving this issue.

peterdavidfagan, 2023-02-23:
> Hi peter, Is there any documentation for building the code in local environment? this will help me build my code before check-in my changes.

@Sujeendra happy to discuss further in case you wish to build locally and contribute to the project.

closed 2023-02-23 by peterdavidfagan

sshaizkhan, 2023-02-23:
@peterdavidfagan I would love to contribute more given my background in robotics and a long history with MoveIt. Let me know. Thanks

sshaizkhan, 2023-02-24:
Hi @peterdavidfagan 
sure you can start tagging me to new issues. Thanks

Sujeendra, 2023-02-27:
HI @peterdavidfagan I was able to setup my local and build it. But, I have one question regarding running unit test cases. How to execute the test cases in local before releasing the code? I saw that there is a folder in moveit_py for running test cases but I get import error when I execute the python code. Can you please help me with this?

peterdavidfagan, 2023-02-27:
> I saw that there is a folder in moveit_py for running test cases but I get import error when I execute the python code. Can you please help me with this?

Thanks for notifying me of this, there is a bug in the unittest code,  they are based on the old library.

peterdavidfagan, 2023-02-28:
https://github.com/ros-planning/moveit2/pull/1979