moveit/moveit2 #1933: Find and remove long-deprecated features

AndyZe, 2023-02-06:
You should be able to find deprecated features like this:

`grep -r deprecated`

Check how long ago they were deprecated and remove if older than, say, 9 months. You can check when a line of code was last modified with `git blame <file_name> -L 80,81` (for example) to check for lines 80-81 in a given file

adityapatruni, 2023-02-06:
I will take this up. Should I follow up on this thread for updates?

AndyZe, 2023-02-06:
Yes please!

AdarshSingh-14, 2023-02-27:
Hey, I would like to help with this issue if someone isn't working on it, and can someone help me get started as I am new to open source.

AndyZe, 2023-02-27:
@adityapatruni are you going to be able to finish your PR? Maybe @AdarshSingh-14 could help you finish it

adityapatruni, 2023-02-27:
@AndyZe Sure that would be great. Sorry I couldn't get to it. I got interrupted by something. @AdarshSingh-14 take a look and let me know how I can assist you. You can use this PR as guide to see which methods can be deprecated completely.

adityapatruni, 2023-02-27:
https://github.com/ros-planning/moveit2/pull/1935

AndyZe, 2023-02-27:
@AdarshSingh-14 I think you should check out @adityapatruni's branch and continue on from there  :+1:

Please try to keep his commit so he gets some credit for his work. Let's continue the discussion here:  https://github.com/ros-planning/moveit2/pull/1935

AdarshSingh-14, 2023-02-28:
I created a fork of moveit2 of mine cloned it locally and then added @adityapatruni repo as remote and checked out to your PR branch. In that way, I can commit to your existing PR.

Hope that's the right approach.

adityapatruni, 2023-02-28:
That’s great Adarsh. This will work well.

AdarshSingh-14, 2023-03-04:
@adityapatruni  I think I require your permission to commit to your PR as it's failing to get pushed.

adityapatruni, 2023-03-04:
Oh is that so. Let me check and give you permission.

adityapatruni, 2023-03-04:
@AdarshSingh-14  if you have write access to ros-planning/moveit2 you should be able to add new commits to my PR. The option to allow commits by maintainers is already enabled. @AndyZe Maybe you can help with this?

AdarshSingh-14, 2023-03-04:
remote: Permission to adityapatruni/moveit2.git denied to AdarshSingh-14.
fatal: unable to access 'https://github.com/adityapatruni/moveit2.git/': The requested URL returned error: 403

How do I fix this?
Or should I create a new PR?

AndyZe, 2023-03-04:
@adityapatruni the easiest way to fix this is go to:  https://github.com/adityapatruni/moveit2/tree/apat-mv_deprecated_features

Then you should be able to click on Settings and give @AdarshSingh-14 write permission.

If you can't do that, then yeah, I think @AdarshSingh-14 will have to open a new PR from his fork. No worries, your commits should still be intact.

adityapatruni, 2023-03-04:
@AdarshSingh-14 You should have received an invite to collaborate

adityapatruni, 2023-03-04:
Let me know if that works. If not create a new PR and tag us there. Cheers!

AdarshSingh-14, 2023-03-04:
@adityapatruni Worked perfectly 🙌.

rmessaou, 2023-09-24:
Hi,
I want to contribute to this issue. As the last PR was closed without being merged and is stale, I created a new PR removing the deprecated old functions here https://github.com/ros-planning/moveit2/pull/2384. 
Would be great if you guys can take a look!

closed 2023-09-29 by tylerjw