moveit/moveit2 #1940: Moveit2 Octomap not visualising

HokulS, 2023-02-09:
When I try to use Moveit2 to display octomaps I do not get any visual.  My initial point cloud has data, but the filtered point cloud has no data.

Humble Binary install
Ubuntu 22.04
Cyclone DDS

Im getting this message from octomap

Listening to '/kuka_lidar/points' using message filter with target frame 'world '

But nothing after that. 
sensors3d.yaml
`sensors:
  - default_sensor
default_sensor:
  filtered_cloud_topic: out_cloud
  max_range: 1000.0
  max_update_rate: 1.0
  padding_offset: 0.1
  padding_scale: 1.0
  point_cloud_topic: /kuka_lidar/points
  point_subsample: 1
  sensor_plugin: occupancy_map_monitor/PointCloudOctomapUpdater '

Launch file
    octomap_config = load_yaml('solar_truck_moveit_config', 'config/sensors_3d.yaml')
    print(octomap_config);

    octomap_updater = {'octomap_frame': 'world',
                       'octomap_resolution': 0.05,
                       'max_range': 1000.0}

    move_group = Node(
        package="moveit_ros_move_group",
        executable="move_group",
        output="screen",
        parameters=[config_dict, octomap_updater, octomap_config],
        )

Full log after launching
ros2 launch solar_truck_moveit_config moveit_rviz.launch.py
[INFO] [launch]: All log files can be found below /home/hokul/.ros/log/2023-02-09-14-17-51-704888-hokul-XPS-15-9520-1392504
[INFO] [launch]: Default logging verbosity is set to INFO
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
{'sensors': ['default_sensor'], 'default_sensor': {'filtered_cloud_topic': 'out_cloud', 'max_range': 1000.0, 'max_update_rate': 1.0, 'padding_offset': 0.1, 'padding_scale': 1.0, 'point_cloud_topic': '/kuka_lidar/points', 'point_subsample': 1, 'sensor_plugin': 'occupancy_map_monitor/PointCloudOctomapUpdater'}}
[INFO] [ruby $(which ign) gazebo-1]: process started with pid [1392505]
[INFO] [static_transform_publisher-2]: process started with pid [1392508]
[INFO] [robot_state_publisher-3]: process started with pid [1392510]
[INFO] [parameter_bridge-4]: process started with pid [1392513]
[INFO] [create-5]: process started with pid [1392515]
[INFO] [move_group-6]: process started with pid [1392517]
[INFO] [static_transform_publisher-7]: process started with pid [1392519]
[INFO] [rviz2-8]: process started with pid [1392521]
[static_transform_publisher-7] [INFO] [1675916271.933802752] [static_transform_publisher0]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-7] from 'world' to 'platform'
[static_transform_publisher-2] [INFO] [1675916271.935518806] [static_transform_publisher_GI955fYKrm61bQMD]: Spinning until stopped - publishing transform
[static_transform_publisher-2] translation: ('0.000000', '0.000000', '0.000000')
[static_transform_publisher-2] rotation: ('0.000000', '0.000000', '0.000000', '1.000000')
[static_transform_publisher-2] from 'kuka_lidar' to 'robot/kuka_rev_two/kuka_lidar'
[create-5] [INFO] [1675916271.937156423] [ros_gz_sim]: Requesting list of world names.
[robot_state_publisher-3] [INFO] [1675916271.959886015] [robot_state_publisher]: got segment kuka
[robot_state_publisher-3] [INFO] [1675916271.959957398] [robot_state_publisher]: got segment kuka_lidar
[robot_state_publisher-3] [INFO] [1675916271.959966069] [robot_state_publisher]: got segment kuka_rev_five
[robot_state_publisher-3] [INFO] [1675916271.959971455] [robot_state_publisher]: got segment kuka_rev_four
[robot_state_publisher-3] [INFO] [1675916271.959976577] [robot_state_publisher]: got segment kuka_rev_one
[robot_state_publisher-3] [INFO] [1675916271.959980364] [robot_state_publisher]: got segment kuka_rev_six
[robot_state_publisher-3] [INFO] [1675916271.959983159] [robot_state_publisher]: got segment kuka_rev_three
[... 165 lines cut ...]
[ruby $(which ign) gazebo-1] [INFO] [1675916273.426183891] [controller_manager]: Setting use_sim_time=True for joint_state_broadcaster to match controller manager (see ros2_control#325 for details)
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] ../src/intel/isl/isl.c:2216: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] libEGL warning: egl: failed to create dri2 screen
[ruby $(which ign) gazebo-1] ../src/intel/isl/isl.c:2216: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
[ruby $(which ign) gazebo-1] [INFO] [1675916274.211598353] [controller_manager]: Configuring controller 'joint_state_broadcaster'
[ruby $(which ign) gazebo-1] [INFO] [1675916274.211688075] [joint_state_broadcaster]: 'joints' or 'interfaces' parameter is empty. All available state interfaces will be published
[ros2-9] Sucessfully loaded controller joint_state_broadcaster into state active
[INFO] [ros2-9]: process has finished cleanly [pid 1392737]
[INFO] [ros2-10]: process started with pid [1393267]
[ruby $(which ign) gazebo-1] [INFO] [1675916275.567990275] [controller_manager]: Loading controller 'kuka_controller'
[ruby $(which ign) gazebo-1] [INFO] [1675916275.573980845] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ruby $(which ign) gazebo-1] [INFO] [1675916275.574006095] [controller_manager]: Setting use_sim_time=True for kuka_controller to match controller manager (see ros2_control#325 for details)
[ruby $(which ign) gazebo-1] [INFO] [1675916275.589693238] [controller_manager]: Configuring controller 'kuka_controller'
[ruby $(which ign) gazebo-1] [INFO] [1675916275.589819237] [kuka_controller]: No specific joint names are used for command interfaces. Using 'joints' parameter.
[ruby $(which ign) gazebo-1] [INFO] [1675916275.589870767] [kuka_controller]: Command interfaces are [position] and state interfaces are [position velocity].
[ruby $(which ign) gazebo-1] [INFO] [1675916275.589890896] [joint_trajectory_controller.interpolation_methods]: No interpolation method parameter was given. Using the default, VARIABLE_DEGREE_SPLINE.
[ruby $(which ign) gazebo-1] [INFO] [1675916275.589894073] [kuka_controller]: Using 'splines' interpolation method.
[ruby $(which ign) gazebo-1] [INFO] [1675916275.590484235] [kuka_controller]: Controller state will be published at 50.00 Hz.
[ruby $(which ign) gazebo-1] [INFO] [1675916275.590898860] [kuka_controller]: Action status changes will be monitored at 20.00 Hz.
[ros2-10] Sucessfully loaded controller kuka_controller into state active
[INFO] [ros2-10]: process has finished cleanly [pid 1393267]
[move_group-6] [INFO] [1675916285.122537810] [move_group]: Message Filter dropping message: frame 'robot/kuka_rev_two/kuka_lidar' at time 0.004 for reason 'the timestamp on the message is earlier than all the data in the transform cache

JaouadROS, 2023-03-01:
Did you solve the bug?
I have the same issue with gazebo on ubuntu 22.04.

HokulS, 2023-03-03:
> Did you solve the bug? I have the same issue with gazebo on ubuntu 22.04.

No I wasn't able to unfortunately

JaouadROS, 2023-03-03:
I solved mine, it was a problem related to CPU/GPU usage. Check [this](https://github.com/gazebosim/docs/blob/master/fortress/troubleshooting.md#problems-with-dual-intel-and-nvidia-gpu-systems), it might help you.

tejasps28, 2023-06-25:
Hi everyone, has anyone been able to find a solution for this?

closed 2024-03-15 by github-actions[bot]