moveit/moveit2 #1942: Segfault in moveit_rviz_plugin::MotionPlanningDisplay::fixedFrameChanged()

wshackle, 2023-02-09:
### Description

Overview of your issue here.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source or Binary build? See instructions below
* If binary, which release version?
* If source, which branch?
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone 

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.
Setup tutorial as described in https://moveit.picknik.ai/humble/doc/tutorials/getting_started/getting_started.html
Start tutorial as described in https://moveit.picknik.ai/humble/doc/tutorials/quickstart_in_rviz/quickstart_in_rviz_tutorial.html
Run the following command in a separate terminal
    
    ros2 launch moveit2_tutorials demo.launch.py rviz_tutorial:=true

(Note rviz2 doesn't start)

Run rviz2 in a new terminal
Add the MotionPlanning visualization
Change the fixed frame under GlobalOptions from map to panda_link0

### Expected behaviour
rviz2 should not crash

### Actual behaviour
Rviz2 has a segfault

### Backtrace or Console output

The console output of rviz2 : 

It starts with:

        shackle@SQUALL:~/ws_moveit2$ rviz2
        Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
        [INFO] [1675960123.345118122] [rviz2]: Stereo is NOT SUPPORTED
        [INFO] [1675960123.345166914] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
        [INFO] [1675960123.356433030] [rviz2]: Stereo is NOT SUPPORTED
        Warning: class_loader.impl: SEVERE WARNING!!! A namespace collision has occurred with plugin factory for class rviz_default_plugins::displays::InteractiveMarkerDisplay. New factory will OVERWRITE existing one. This situation occurs when libraries containing plugins are directly linked against an executable (the one running right now generating this message). Please separate plugins out into their own library or just don't link against the library and use either class_loader::ClassLoader/MultiLibraryClassLoader to open.
                 at line 253 in /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp
        [ERROR] [1675960147.410762839] [moveit_ros_visualization.motion_planning_frame]: Action server: /recognize_objects not available
        [INFO] [1675960147.431291771] [moveit_ros_visualization.motion_planning_frame]: MoveGroup namespace changed: / -> . Reloading params.
        [INFO] [1675960147.775675969] [moveit_rdf_loader.rdf_loader]: Loaded robot model in 0.321374 seconds
        [INFO] [1675960147.775752140] [moveit_robot_model.robot_model]: Loading robot model 'panda'...
        [WARN] [1675960147.791543145] [moveit_ros.robot_model_loader]: No kinematics plugins defined. Fill and load kinematics.yaml!
        [INFO] [1675960147.806967128] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Starting planning scene monitor
        [INFO] [1675960147.807621716] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Listening to '/monitored_planning_scene'
        [INFO] [1675960147.983762700] [interactive_marker_display_93940753848528]: Connected on namespace: /rviz_moveit_motion_planning_display/robot_interaction_interactive_marker_topic
        [INFO] [1675960147.985190855] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
        [INFO] [1675960147.985556628] [moveit_ros_robot_interaction.robot_interaction]: No active joints or end effectors found for group 'hand'. Make sure that kinematics.yaml is loaded in this node's namespace.
        Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
                 at line 93 in ./src/buffer_core.cpp
        [INFO] [1675960147.988286085] [moveit_ros_visualization.motion_planning_frame]: group hand
        [INFO] [1675960147.988297438] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'hand' in namespace ''
        [INFO] [1675960147.991850279] [move_group_interface]: Ready to take commands for planning group hand.
        [INFO] [1675960147.992332211] [moveit_ros_visualization.motion_planning_frame]: group hand
 
Then repeats many times:

        [INFO] [1675960147.993276277] [interactive_marker_display_93940753848528]: Sending request for interactive markers
        Warning: Invalid frame ID "map" passed to canTransform argument target_frame - frame does not exist
                 at line 93 in ./src/buffer_core.cpp

Then ends with:

[... 202 lines cut ...]
        #13 0x00007efe09b154c6 in QComboBoxPrivateContainer::itemSelected(QModelIndex const&) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #14 0x00007efe09b15df0 in QComboBoxPrivateContainer::eventFilter(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #15 0x00007efe095fbb9a in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #16 0x00007efe09a0d702 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #17 0x00007efe09a15364 in QApplication::notify(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #18 0x00007efe095fbe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #19 0x00007efe09a13e47 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #20 0x00007efe09a6a843 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #21 0x00007efe09a6cfd5 in  () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #22 0x00007efe09a0d713 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Widgets.so.5
        #23 0x00007efe095fbe3a in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #24 0x00007efe082a4307 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
        #25 0x00007efe08279a2c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Gui.so.5
        #26 0x00007efe03001d6e in  () at /lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
        #27 0x00007efe07890d3b in g_main_context_dispatch () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
        #28 0x00007efe078e56c8 in  () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
        #29 0x00007efe0788e3e3 in g_main_context_iteration () at /lib/x86_64-linux-gnu/libglib-2.0.so.0
        #30 0x00007efe096550b8 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #31 0x00007efe095fa75b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #32 0x00007efe09602cf4 in QCoreApplication::exec() () at /lib/x86_64-linux-gnu/libQt5Core.so.5
        #33 0x0000559aef07199e in  ()
        #34 0x00007efe08cecd90 in __libc_start_call_main (main=main@entry=0x559aef071430, argc=argc@entry=1, argv=argv@entry=0x7ffc13b9dba8) at ../sysdeps/nptl/libc_start_call_main.h:58
        #35 0x00007efe08cece40 in __libc_start_main_impl (main=0x559aef071430, argc=1, argv=0x7ffc13b9dba8, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffc13b9db98)
            at ../csu/libc-start.c:392
        #36 0x0000559aef071c15 in  ()

breezZe, 2023-03-10:
I have the same issue.

breezZe, 2023-03-10:
@wshackle I've got a method to work around this. 
Set ```fixed_frame``` to ```/panda_link0``` as soon as RVIZ2 started, then launch moveit tutorial node will work. RVIZ2 will not crash. Maybe you can try it. I still don't know the reason, however I think it is a rviz2 problem but not moveit?