moveit/moveit2 #1944: MoveIt Setup Assistant not working!

vidalperezbohoyo, 2023-02-12:
### Description
I'm trying to setup a new robot, but I'm barely able to get past the first step. The program stops responding when clicking on create a new package and trying to load an urdf. Even writing something in the textbox already stops working. It doesn't get any messages from the terminal.

### Your environment
* ROS Distro: Humble
* OS Version: 22.04
* Source and binary (same with this error) 
* Latest
* Latest
* Cyclone DDS

### Steps to reproduce
If write on textbox in create new package import urdf or browse and select any urdf.

patricia241, 2023-02-15:
I have the same problem.

LSBigum, 2023-02-18:
I haven't got this issue. Did the following:
```sudo apt install ros-humble-moveit``` 
Sourced the underlay and my overlay (```source /opt/ros/humble/setup.bash``` and ```source ros2_ws/install/setup.bash```)
Ran ```ros2 launch moveit_setup_assistant setup_assistant.launch.py verbose:=true```
And then I was able to load my .xacro file and generate a moveit config.

- ROS Distro: Humble
- OS Version: 22.04
- Binary install
- Fast DDS

vidalperezbohoyo, 2023-02-18:
Doesn't work. I have this installed. With verbose nothing is displayed when crashes.

vidalperezbohoyo, 2023-02-18:
I have tried to unistall it , then autoremove to clean all. Then changed Ubuntu to US lenguage and then reboot. Then installed again with your command. But is NOT working...

vidalperezbohoyo, 2023-02-19:
I have retried in a virtualbox from zero. Same error. 

Now there is more info:

```
[INFO] [launch]: All log files can be found below /home/vidi/.ros/log/2023-02-19-14-07-51-890511-vidi-VirtualBox-53720
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [moveit_setup_assistant-1]: process started with pid [53721]
[moveit_setup_assistant-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[ERROR] [moveit_setup_assistant-1]: process has died [pid 53721, exit code -9, cmd '/opt/ros/humble/lib/moveit_setup_assistant/moveit_setup_assistant --ros-args'].

```

Is this version of moveit. Assistant is broken!

mikeferguson, 2023-02-20:
I just installed humble debs to test this out - could not reproduce the error as noted here - the things I did find:

 * https://github.com/ros-planning/moveit2/pull/1962
 * If you select a urdf that is not actually on your ROS2 path - it crashes due to ament_index_cpp::PackageNotFoundError in setPackageName() function

Since I noticed you have:
```
[moveit_setup_assistant-1] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
```
I'm thinking that this is due to some aspect of your Ubuntu/X11 setup. On my machine (which is basically a plain desktop 22.04 install), I have XDG_SESSION_TYPE=x11. Have you tried setting QT_QPA_PLATFORM to wayland as it suggests?

vidalperezbohoyo, 2023-02-21:
Thanks! I found the problem: urdf must be inside a package in the workspace and sourced.  Also there is a bug when you try to type in the textbox. And program should drop an error when urdf is not valid because of the location.  In my opinion the program should not crash for x reasons. Thanks a lot!.

I don`t know if the XDG_SESSION_TYPE affects. I have not changed anything about it. I just installed ubuntu from scratch and this is how it comes.

mikeferguson, 2023-02-21:
It's interesting that you didn't get a giant screen dump error trace - cause I got a mountain of debugging information when I tried to open a URDF not on the path - I'm going to change the title of this ticket and assign to myself to track down a fix for the urdf issue

mikeferguson, 2023-02-21:
WIP is over here: https://github.com/mikeferguson/moveit2/tree/fix_1944 - this fixes the first issue (typing into the box doesn't crash) and starts to fix the second (not on ROS package path no longer crashes, but we end up in an odd state and try to load various non-existent meshes - still gotta fix that)

closed 2023-04-01 by AndyZe