moveit/moveit2 #1951: pick() and place() functions in moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp commented

atuldivekar, 2023-02-16:
### Description

pick() and place() functions in moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp 
are commented which causes code to fail.

There is no documentation to indicate this is deprecated and tutorials such as 
  https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/pick_place/src/pick_place_tutorial.cpp
still use these functions

* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Binary build for ROS2 Humble , source build for MoveIt2 Humble branch
Cyclone DDS

### Steps to reproduce

I tried to create a simple pick and place demo.

The relevant lines are : 
 
#include <moveit/planning_scene_interface/planning_scene_interface.h>
#include <moveit/move_group_interface/move_group_interface.h>

...

move_group_interface_ptr_     = new MoveGroupInterface(node_,"panda_arm");
move_group_interface_ptr_->pick("object", grasps);   

Compiler error:
>>  error: ‘class moveit::planning_interface::MoveGroupInterface’ has no member named ‘pick’

The code for pick() etc. appears to be commented in 
moveit_ros/planning_interface/move_group_interface/src/move_group_interface.cpp

If this is deprecated, this should be documented and the tutorials updated.
If deprecated, what is the alternative to use? MoveIt Task Constructor?

DaniGarciaLopez, 2023-03-04:
I am also interested to know why these functions are commented out. If they are deprecated in Humble I would like to find a simple alternative to pick up objects from the planning scene without the use of Moveit Task Constructor. Thanks!

henningkayser, 2023-03-10:
@atuldivekar the tutorial states that it is deprecated and, the pick place capability has been removed a while ago (https://github.com/ros-planning/moveit2/pull/1177). However, you are right that commented code in the main repo is not great. Would you mind filing a PR to remove it along with the tutorial?

And yes, MoveIt Task Constructor is really the best solution for pick and place tasks.https://moveit.picknik.ai/humble/doc/tutorials/pick_and_place_with_moveit_task_constructor/pick_and_place_with_moveit_task_constructor.html

closed 2023-06-15 by tylerjw

antonio1matos, 2024-04-22:
@DaniGarciaLopez  did you found a way to pick up objects from the planning scene avoiding the use of Moveit Task Constructor?

DaniGarciaLopez, 2024-04-22:
@antonio1matos We decided to use MoveIt Task Constructor. It has many introspection tools and it's worth a try for simple pick/place tasks as well.

PR #2239 (2023-06-15): Remove deprecated move_group pick& place functionality + demo
### Description

Fixes #1951 

### Checklist
- [ ] **Required by CI**: Code is auto formatted using [clang-format](http://moveit.ros.org/documentation/contributing/code)
- [ ] Extend the tutorials / documentation [reference](http://moveit.ros.org/documentation/contributing/)
- [ ] Document API changes relevant to the user in the [MIGRATION.md](https://github.com/ros-planning/moveit2/blob/main/MIGRATION.md) notes
- [ ] Create tests, which fail without this PR [reference](https://moveit.picknik.ai/humble/doc/examples/tests/tests_tutor [... cut]