moveit/moveit2 #1953: moveit_task_constructor_core failed to build

DolceParadise, 2023-02-17:
### Description

Build for MoveIt2 Tutorials fails on ROS Rolling

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Source or Binary build? Binary
* If binary, which release version? 2.6.0-1jammy.20230131.064416
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone DDS

### Steps to reproduce
Follow along with the tutorial steps at https://moveit.picknik.ai/main/doc/tutorials/getting_started/getting_started.html 

### Expected behaviour
Build completes successfully and the message "Summary: 54 packages finished" appears. 

### Actual behaviour
The build ends with the following output:
```
error: ‘struct planning_interface::MotionPlanResponse’ has no member named ‘trajectory_’; did you mean ‘trajectory’?
  208 |         result = res.trajectory_;
      |                      ^~~~~~~~~~~
      |                      trajectory
gmake[2]: *** [src/CMakeFiles/moveit_task_constructor_core.dir/build.make:244: src/CMakeFiles/moveit_task_constructor_core.dir/solvers/pipeline_planner.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:254: src/CMakeFiles/moveit_task_constructor_core.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_task_constructor_core [7.73s, exited with code 2]

Summary: 21 packages finished [13.1s]
  1 package failed: moveit_task_constructor_core
  2 packages had stderr output: moveit_configs_utils moveit_task_constructor_core
  32 packages not processed
```

JafarAbdi, 2023-02-18:
Should be fixed with the latest moveit_task_constructor's ros2 branch

closed 2023-02-18 by JafarAbdi

DolceParadise, 2023-02-19:
Had a look at this [PR](https://github.com/ros-planning/moveit_task_constructor/pull/426) and got what was breaking the build. Thanks a lot!