moveit/moveit2 #1959: Very hard to simulate with Gazebo with the controlers of the moveit2 config

FrGe2016, 2023-02-19:
### Description
Moveit 1 had a good tutorial for the integration of gazebo and a setup assistant for the simulation.

https://ros-planning.github.io/moveit_tutorials/doc/gazebo_simulation/gazebo_simulation.html

Since i am working with movit2 and humble. After multiple google searchs, i  have not been able to find the same ressources and have not succeed to make movit2 working along gazebo.

Overview of your issue here.

### Your environment
* ROS humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.

sjahr, 2023-02-20:
Yes, there is no moveit2 gazebo tutorial available yet. I know that @AndrejOrsula wrote a helpful demo package to show how MoveIt2 and Gazebo Ignition used together: https://github.com/AndrejOrsula/panda_ign_moveit2. Furthermore, there is an open PR under review in the moveit2_tutorials that enables using the Isaac Simulator with Moveit2 https://github.com/ros-planning/moveit2_tutorials/pull/590

If you have some time to spare, we would appreciate it if you make a contribution to https://github.com/ros-planning/moveit2_tutorials to contribute instructions on how to set up MoveIt 2 with Gazebo Ignition. You could either port the existing one from moveit1 or write a completely new one based on your experience with getting Gazebo to work with moveit2.

AndyZe, 2023-02-20:
If you're looking for Gazebo classic, there's this:  https://github.com/UniversalRobots/Universal_Robots_ROS2_Gazebo_Simulation

But it looks like it might be a little out of date

closed 2023-05-22 by github-actions[bot]