moveit/moveit2 #1970: CI is broken on Rolling

mikeferguson, 2023-02-23:
The fix is almost certainly needed to be upstream - just ticketing here for visibility (so we don't have multiple people working on it)

 * @AndyZe tried to fix this with https://github.com/ros-planning/moveit_msgs/pull/153 - BUT:
 * service_msgs was added only to Rolling

It looks like the failure is actually in sensor_msgs:

[image]

mikeferguson, 2023-02-23:
Locally, I can build moveit_msgs against ros2-testing debs (and I uninstalled all of ros-rolling-* before tasting, so it's not a missing dependency).

sjahr, 2023-03-06:
This commit could be the culprit: https://github.com/ros2/rosidl_defaults/commit/a3f753d1527cbc8be3b6c41c172d68732414584d. But it still does not explain why service_msgs is not installed :thinking:

jspricke, 2023-03-06:
The job uses an old Docker image of the ros2-testing repos where ros-rolling-service-msgs is not installed and it does not `apt dist-upgrade` the system: https://github.com/ros-planning/moveit2/blob/main/.docker/source/Dockerfile#L28. So service_msgs is never picked up and thus missing. The logical fix is to use dist-upgrade in the job, as proposed in a144a3c but Docker lint complains about that: https://github.com/jspricke/moveit2/actions/runs/4346334681. So you can either fix Docker lint or build a new Docker base image and run into this problem again, next time. Your call.

mikeferguson, 2023-03-06:
@tylerjw - I think this may be more up your alley - since it looks like it is our CI, not some upstream break

jspricke, 2023-03-07:
Discussed this in the standup, the problem is that the ci-testing Docker images are based on the ci images which the added ros2-testing apt sources: https://github.com/ros-planning/moveit2/blob/main/.docker/ci-testing/Dockerfile and the `apt-get upgrade` in there is not enough to install a new dependency package. @henningkayser will rework the Dockerfile to work from a base image.

vatanaksoytezer, 2023-03-07:
I think we only need to add a rosdep step to ci-testing and that should be sufficient. The dependency in question (service_msgs) is not being installed in apt update / apt upgrade, it should be installed and triggered through `rosdep install`

jspricke, 2023-03-07:
@vatanaksoytezer no, service_msgs is a dependency of rcl already.