moveit/moveit2 #1971: Setup assistant crashed after trying to zoom/pan, rotate the robot model and pressing any keyboard key.

sshaizkhan, 2023-02-23:
### Description

Failed `zoom` or `rotate` the robot model in the Rviz view right pane. Also, once the rviz view pane is activated by hovering cursor or pressing lef/right button on the mouse and if any of the key on the keyboard is pressed, setup assistant will crash.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04.2 LTS on Docker v23.0.1 build a5ee5b1
* Source or Binary build? -> Source
* If source, which branch? -> [humble](https://github.com/ros-planning/moveit2/tree/humble)
* Which RMW (Fast DDS or Cyclone DDS)? -> Cyclone DDS

### Steps to reproduce
1. Launch moveit setup assistant `ros2 launch moveit_setup_assistant setup_assistant.launch.py `.
2. You can try to create new moveit config package or edit an existing one. To check it quickly, press edit existing  moveit config and upload the panda_moveit_config. If you don't have the repo, clone it from [here](https://github.com/ros-planning/moveit_resources/tree/humble/panda_moveit_config).
3. Once the package is uploaded, the robot model will appear on the right. Click anywhere inside the Rviz pane and try to Zoom/Pan, rotate and it will fail to do so.
4. Now if any of the button is pressed on keyboard, the setup assistant will crash with `Segmentation fault (Address not mapped to object [(nil)])`.

### Expected behaviour
Should be able to Zoom/Pan, Rotate the robot model.

### Actual behaviour
Setup Assistant crashed with `Segmentation fault (Address not mapped to object [(nil)])`

### Backtrace or Console output

```
[moveit_setup_assistant-1] Stack trace (most recent call last):
[moveit_setup_assistant-1] #21   Object "", at 0xffffffffffffffff, in 
[moveit_setup_assistant-1] #20   Object "/home/bot/moveit2_ws/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant", at 0x5616e64dc194, in _start
[moveit_setup_assistant-1] #19   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f579c3f9e3f, in __libc_start_main
[moveit_setup_assistant-1] #18   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7f579c3f9d8f, in 
[moveit_setup_assistant-1] #17   Object "/home/bot/moveit2_ws/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant", at 0x5616e64dbda4, in main
[moveit_setup_assistant-1] #16   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f579cb68cf3, in QCoreApplication::exec()
[moveit_setup_assistant-1] #15   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f579cb6075a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
[moveit_setup_assistant-1] #14   Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f579cbbb0b7, in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
[moveit_setup_assistant-1] #13   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f579b1873e2, in g_main_context_iteration
[moveit_setup_assistant-1] #12   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f579b1de6c7, in 
[moveit_setup_assistant-1] #11   Object "/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0.7200.4", at 0x7f579b189d3a, in g_main_context_dispatch
[moveit_setup_assistant-1] #10   Object "/usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5.15.3", at 0x7f5790791d6d, in 
[moveit_setup_assistant-1] #9    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7f579cf1da2b, in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>)
[moveit_setup_assistant-1] #8    Object "/usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3", at 0x7f579cf444b9, in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*)
[moveit_setup_assistant-1] #7    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f579cb61e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[moveit_setup_assistant-1] #6    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f579d651712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[moveit_setup_assistant-1] #5    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f579d6b102e, in 
[moveit_setup_assistant-1] #4    Object "/usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3", at 0x7f579cb61e39, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
[moveit_setup_assistant-1] #3    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f579d65949d, in QApplication::notify(QObject*, QEvent*)
[moveit_setup_assistant-1] #2    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f579d651712, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
[moveit_setup_assistant-1] #1    Object "/usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3", at 0x7f579d695377, in QWidget::event(QEvent*)
[moveit_setup_assistant-1] #0    Object "/opt/ros/humble/lib/librviz_common.so", at 0x7f579dea194f, in rviz_common::ToolManager::handleChar(QKeyEvent*, rviz_common::RenderPanel*)
[moveit_setup_assistant-1] Segmentation fault (Address not mapped to object [(nil)])
[ERROR] [moveit_setup_assistant-1]: process has died [pid 2467, exit code -11, cmd '/home/bot/moveit2_ws/install/moveit_setup_assistant/lib/moveit_setup_assistant/moveit_setup_assistant --ros-args'].
```

@DLu I'm tagging you assuming you were the last person to made changes. Let me know if you need any more information.

vidalperezbohoyo, 2023-05-05:
I have the same problem...

UgurYakut, 2023-10-08:
same issue on local

commit 38668bf1 (2023-10-11): Fix #1971 (#2428)
commit e538316a (2023-10-24): Fix #1971 (#2428) (#2430)

(cherry picked from commit 38668bf1a0c26bd0d972438f6e707f111f52270c)
commit fe0b5219 (2023-10-24): Fix MSA RViz Panel pan/zoom (backport #2428) (#2431)

Fixes #1971

(cherry picked from commit 38668bf1a0c26bd0d972438f6e707f111f52270c)
closed 2023-10-11 by tylerjw