moveit/moveit2 #1973: Executing planned trajectory of planned path fails

HenningLF, 2023-02-24:
### Description
Executing planned trajectory of planned path fails. 
NB: I will be away for a couple of days after posting.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source, Humble 
* Cyclone DDS.

### Background
I have imported my custom robot using the MoveIt wizard. When running the demo.launch file, it works mostly fine. The dimensions and limits works as expected (in Rviz), and when I press _Plan_ the planning is completed, and I can see the animation.

### What is wrong?ssssssssssssssss
1. Executing the plan fails (see console output below)
2. I cannot move the interactive marker. To be more specific, it snaps back when I release the mouse.
3. The robot workspace (Show workspace in Rviz) shows a 2x2x2 box with (0,0,0) in its center. I suspect that this the ill-defined workspace might be a central cause of my problems.
[image]

### What have I tried?
1. Inspecting the URDF, made sure that I do not have any effort velocity limits = 0.
2. Added safety-controllers to the URDF. Suspected that some limits were set to 0 by default, and hence affected the workspace.
3. Loads of research to no avail.   

### Console output
```
[gdb-3] [INFO] [1677248896.098463613] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[gdb-3] [INFO] [1677248896.098586097] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[gdb-3] [INFO] [1677248896.098631454] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[gdb-3] [INFO] [1677248896.098711163] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[gdb-4] [INFO] [1677248896.098709735] [move_group_interface]: Execute request accepted
[gdb-3] [INFO] [1677248896.098761587] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[gdb-3] [INFO] [1677248897.098869380] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[gdb-3] Check clock synchronization if your are running ROS across multiple machines!
[gdb-3] [WARN] [1677248897.098909710] [moveit_ros.trajectory_execution_manager]: Failed to validate trajectory: couldn't receive full current joint state within 1s
[gdb-3] [INFO] [1677248897.098937671] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[gdb-4] [INFO] [1677248897.099313163] [move_group_interface]: Execute request aborted
[gdb-4] [ERROR] [1677248897.199106255] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
[gdb-4] [INFO] [1677248902.871705731] [moveit_ros_visualization.motion_planning_frame]: group amv_4200_shotcrete_arm
[gdb-4] [INFO] [1677248902.871750982] [moveit_ros_visualization.motion_planning_frame]: Constructing new MoveGroup connection for group 'amv_4200_shotcrete_arm' in namespace ''
[gdb-4] [INFO] [1677248902.879561538] [move_group_interface]: Ready to take commands for planning group amv_4200_shotcrete_arm.
[gdb-4] [INFO] [1677248904.883667526] [move_group_interface]: MoveGroup action client/server ready
[gdb-3] [INFO] [1677248904.884534067] [moveit_move_group_default_capabilities.move_action_capability]: Received request
[gdb-3] [INFO] [1677248904.884851975] [moveit_move_group_default_capabilities.move_action_capability]: executing..
[gdb-4] [INFO] [1677248904.885072076] [move_group_interface]: Planning request accepted
[gdb-3] [INFO] [1677248905.884992192] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[gdb-3] Check clock synchronization if your are running ROS across multiple machines!
[gdb-3] [WARN] [1677248905.885028967] [moveit_ros.planning_scene_monitor.planning_scene_monitor]: Failed to fetch current robot state.
```

AndyZe, 2023-02-24:
I'm pretty sure this is the root cause of the issue:

```
[gdb-3] [INFO] [1677248897.098869380] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.
[gdb-3] Check clock synchronization if your are running ROS across multiple machines!
```

We need to figure out why your `joint_state_publisher` isn't working. This is something related to ros2_control.

If you do `ros2 topic echo /joint_states`, do you see any messages?

AndyZe, 2023-02-24:
Here are some references to compare against:

https://github.com/ros-planning/moveit_resources/blob/2c9c6a8f22c45398f55d74b1a3954db1d408cb33/panda_moveit_config/config/ros2_controllers.yaml#L12

https://github.com/ros-planning/moveit_resources/blob/ros2/panda_moveit_config/config/panda.ros2_control.xacro

HenningLF, 2023-02-27:
Thank you for helping!

`ros2 topic echo /joint_states` does not return anything when I press execute, or anything else I have tried.

My ros2_controllers.yaml, among other things, is missing the `command_interface` and `_state_interface_` parameters:
```
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    AMV_4200_Shotcrete_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

AMV_4200_Shotcrete_arm_controller:
  ros__parameters:
    joints:
      - ptSlewing
      - ptJib
      - ptBoom2
      - ptBoom3
      - ptRotate
      - ptTilit
    command_interfaces:
      []
    state_interfaces:
      []
```

I have not included the `ros2_control` in my xacro/urdf, and I guess this is where the root to my problem. Should I write this tag manually, or can the wizzard help me?

HenningLF, 2023-02-27:
## Update:

I regenerated the `ros2_controllers.yaml`. It was not up-to-date with my move groups.
<details>
<summary>Updated ros2_controllers.yaml </summary>

```yaml
# This config file is used by ros2_control
controller_manager:
  ros__parameters:
    update_rate: 100  # Hz

    AMV_4200_Shotcrete_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    amv_4200_shotcrete_arm_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    amv_4200_shotcrete_tool_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

AMV_4200_Shotcrete_arm_controller:
  ros__parameters:
    joints:
      - ptSlewing
      - ptJib
      - ptBoom2
      - ptBoom3
      - ptRotate
      - ptTilit
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
amv_4200_shotcrete_arm_controller:
  ros__parameters:
    joints:
      - ptSlewing
      - ptJib
      - ptBoom2
      - ptBoom3
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
amv_4200_shotcrete_tool_controller:
  ros__parameters:
    joints:
      - ptRotate
      - ptTilit
    command_interfaces:
      - position
    state_interfaces:
      - position
      - velocity
```

</details>

After this change the `ros2 topic echo /joint_states` command started to work. I can see the joint states being published.
<details> 
<summary>Output of ros2 topic echo /joint_states</summary>

```
header:
stamp:
    sec: 1677491495
    nanosec: 516268558
frame_id: ''
name:
- ptJib
- ptSlewing
- ptBoom3
- ptRotate
- ptBoom2
- ptTilit
position:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
velocity:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
effort:
- .nan
- .nan
- .nan
- .nan
- .nan
- .nan
---
```
</details>

In Rviz the problems remains as before. The workspace is wrong, and I cannot execute trajectories or use the interactive markers. However, I now get a **different output in the demo.launch.py in the terminal**. 
<details> <summary>Output from demo.launch.py</summary>

```
[move_group-3] [INFO] [1677491876.926441107] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Received goal request
[move_group-3] [INFO] [1677491876.926622597] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution request received
[move_group-3] [INFO] [1677491876.926677006] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1677491876.926689343] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1677491876.926776931] [moveit_ros.trajectory_execution_manager]: Validating trajectory with allowed_start_tolerance 0.01
[rviz2-4] [INFO] [1677491876.926643151] [move_group_interface]: Execute request accepted
[move_group-3] [INFO] [1677491876.930578929] [moveit_ros.trajectory_execution_manager]: Starting trajectory execution ...
[move_group-3] [INFO] [1677491876.930640058] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1677491876.930652031] [moveit.plugins.moveit_simple_controller_manager]: Returned 1 controllers in list
[move_group-3] [INFO] [1677491876.930771489] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: sending trajectory to AMV_4200_Shotcrete_arm_controller
[ros2_control_node-5] [INFO] [1677491876.930914697] [AMV_4200_Shotcrete_arm_controller]: Received new action goal
[ros2_control_node-5] [ERROR] [1677491876.930947885] [AMV_4200_Shotcrete_arm_controller]: Joints on incoming trajectory don't match the controller joints.
[move_group-3] [INFO] [1677491876.931037232] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: AMV_4200_Shotcrete_arm_controller started execution
[move_group-3] [WARN] [1677491876.931051403] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal request rejected
[move_group-3] [ERROR] [1677491876.931038148] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Goal was rejected by server
[move_group-3] [ERROR] [1677491876.931067380] [moveit_ros.trajectory_execution_manager]: Failed to send trajectory part 1 of 1 to controller AMV_4200_Shotcrete_arm_controller
[move_group-3] [INFO] [1677491876.931071619] [moveit_ros.trajectory_execution_manager]: Completed trajectory execution with status ABORTED ...
[move_group-3] [INFO] [1677491876.931124867] [moveit_move_group_default_capabilities.execute_trajectory_action_capability]: Execution completed: ABORTED
[rviz2-4] [INFO] [1677491876.931301653] [move_group_interface]: Execute request aborted
[rviz2-4] [ERROR] [1677491877.026369415] [move_group_interface]: MoveGroupInterface::execute() failed or timeout reached
```
</details>

HenningLF, 2023-02-27:
## Non-deterministic error messages
I have noticed something interesting. The planning and execution sometimes work now, and sometimes not, without making **any** changes. Whenever the execution fails, I can get different error messages.
- `[move_group-3] [INFO] [moveit_ros.current_state_monitor]: Didn't received robot state (joint angles) with recent timestamp within 1.000000 seconds.`
- [ros2_control_node-5] [ERROR] [amv_4200_shotcrete_arm_controller]: Can't accept new action goals. Controller is not running.
- [move_group-3] [ERROR] [moveit.simple_controller_manager.follow_joint_trajectory_controller_handle]: Action client not connected to action server: amv_4200_shotcrete_arm_controller/follow_joint_trajectory

I noticed some other error messages at startup that also were non-deterministic. 
```
[spawner-6] 1677504382.567353 [87]    spawner: Failed to find a free participant index for domain 87
[spawner-6] [ERROR] [rmw_cyclonedds_cpp]: rmw_create_node: failed to create domain, error Error
[spawner-6] 
[spawner-6] >>> [rcutils|error_handling.c:108] rcutils_set_error_state()
[spawner-6] This error state is being overwritten:
[spawner-6] 
[spawner-6]   'error not set, at ./src/rcl/node.c:263'
[spawner-6] 
[spawner-6] with this new error message:
[spawner-6] 
[spawner-6]   'rcl node's rmw handle is invalid, at ./src/rcl/node.c:415'
[spawner-6] 
[spawner-6] rcutils_reset_error() should be called after error handling to avoid this.
[spawner-6] <<<
[spawner-6] [ERROR] [rcl]: Failed to fini publisher for node: 1
```

Followed by
`[ERROR] [spawner-6]: process has died [pid 513370, exit code 1, cmd '/opt/ros/humble/lib/controller_manager/spawner AMV_4200_Shotcrete_arm_controller --ros-args'].`

Which spawners that die, 6,7,8 and/or 9, vary between each run. It seems like the which error message I get when pressing execute (plan) is determined by which spawner that dies at startup. 

I cannot figure out why the spawners die or how to fix it. I suspect that the following warning could be relevant:

AndyZe, 2023-02-27:
Are you using Cylcone or FastDDS? The MoveIt installation instructions recommend Cyclone b/c of issues like this

HenningLF, 2023-02-27:
> Are you using Cylcone or FastDDS? The MoveIt installation instructions recommend Cyclone b/c of issues like this

I'm using Cylcone as recommended, think it should be correctly installed. 
Running: `echo $RMW_IMPLEMENTATION`
Returns: `rmw_cyclonedds_cpp`

AndyZe, 2023-02-27:
OK. Well, this sounds like a Cyclone configuration issue, to me. I'd probably raise an issue here:  https://github.com/eclipse-cyclonedds/cyclonedds

You could also try switching back to FastDDS since it's quick and easy to try.

Also, if you're testing in a public place like a coffee shop, it can behave differently than at your home network. Sorry that this is all I can recommend.

HenningLF, 2023-02-27:
Thank you, I will follow up your suggestions tomorrow.

closed 2023-02-28 by HenningLF