moveit/moveit2 #1975: Using moveit_servo, how to obtain the real-time speed in both Cartesian space and Joint space

KingWang569, 2023-02-26:
### Description

I successfully run moveit_servo on ROS1 melodic and ROS2 humable.
But I have a problem: I can't know the exactly the cartesian velocity and the joint velocity of the manipulator running in Gazebo.
For example, I send the twist command to control the manipulator to run at high speed in the Z direction of Cartesian space, but it may be limited by the maximum speed of a certain joint, and the actual speed in Cartesian space cannot reach the command I sent.
How can I know the current real-time joint velocity and Cartesian space velocity at this time to ensure that my control does not exceed the joint limit？
I can't get it from both /joint_state and /joint_group_position_controller/command topics.

### Your environment
* ROS Distro: ROS1 melodic, ROS2 humable
* OS Version: Ubuntu 18.04, Ubuntu 22.04
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.

### Expected behaviour
There is a topic to see the actual running speed of the manipulator in Cartesian space and joint space

### Actual behaviour
There is no topic to see the actual running speed of the manipulator in Cartesian space and joint space

Pomorondza, 2023-05-07:
I believe that joint velocities you can obtain from /joint_states topic, and Cartesian velocity you can calculate yourself using t2_ros::Buffer and tf2_ros::TransformListener by calulating the difference between two transforms divided by delta time.

ibrahiminfinite, 2023-08-15:
@sjahr 
This can also be closed.

I don't think we will be adding an API for this into Servo, since that would mean having more computations and extra variables to keep track. Also doing it either using tf2 or Jacobin from within Servo will have significant hit on the real-time performance.

closed 2023-08-15 by sjahr