moveit/moveit2 #1977: [Moveit Servo] Is it possible to use multiple controllers?

nicohmje, 2023-02-27:
### Description

I'm trying to implement Moveit2 Servo in a robotic arm that, because of it's actuation, requires two different ros2_control controllers for operation. However, when trying to use Moveit2 Servo, the parameter file only has one "command out topic" (JointTrajectoryController). Is it possible to use two? Can I pipe it through the move group somehow? Thank you.

### Your environment
* ROS Distro: [Humble]
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Binary
* If binary, which release version? Latest
* Which RMW (Fast DDS or Cyclone DDS)? Cyclone DDS

### Steps to reproduce

-------

### Expected behaviour

Moveit2 Servo controls the robot like Moveit2 controls the robot, by sending the different joint commands to their respective controller.

### Actual behaviour
Only one controller can be used, resulting in "Incoming joint trajectory doesn't match joints"

AndyZe, 2023-02-27:
Hi @nicohmje, I don't think that's possible in a tidy way. The best alternative I can think of would be:

- Have Servo publish to a single topic like it usually does
- Have a relay node subscribe to that^ topic and republish it on the 2 different controller topics

schornakj, 2023-03-01:
> Have a relay node subscribe to that^ topic and republish it on the 2 different controller topics

Another way to get an equivalent result is to remap the subscriptions of the two controller topics so they're both listening to the topic published by Servo.

gavanderhoorn, 2023-03-06:
Is this a cross-post of [ros2_control different command interfaces for different joints](https://answers.ros.org/question/412903) on ROS Answers?

ibrahiminfinite, 2023-08-15:
@AndyZe 
Can this issue be closed ?

closed 2023-08-15 by AndyZe