moveit/moveit2 #1981: [Moveit Servo] JoyToServoPub node keeps referencing panda arm, even though servo params specify otherwise.

nicohmje, 2023-03-01:
### Description

Hello,

While trying to implement Joystick teleop using moveit servo on a robot arm, I keep encountering the issue of the nodes making references to the Panda Robot. 

When launching a composable node containing JoyToServoPub and Joy, the servo node errors out saying "panda_link0" wasn't found in my model. I have a parameters yaml file parsed to the servo node, but don't know how to send these parameters to the JoyToServoPub node. 

Should I not be launching moveit_servo::JoyToServoPub? Is there another piece of documentation / example I've missed?

Thank you,

Nicolas Hammje

Ros2 Humble, Ubuntu 22.04, latest Moveit2 binary build.

### Steps to reproduce

Launch a composable node container with JoyToServoPub and Joy. JoyToServoPub publishes delta twist cmds in reference to the frame "panda_link0". 

### Expected behaviour

The servo params specified in the yaml file parsed to the servo node should apply to the whole servo framework. 

### Actual behaviour

The parameters are ignored, and defaults are instead used.

okvik, 2023-03-28:
I just encountered this issue in my experiments with `moveit_servo`. You'll need to modify the code of the `JoyToServoPub` node since it hardcodes various "panda_*" names and frames specific to the demo launches. Same goes for keyboard input node.

Probably some of this code could be parameterized to make it easier for people to do initial experiments on their own robots, but ultimately you'll end up wanting to fork this code and customize it for your exact needs so I don't think *too* much generalization would be useful or wanted here.

ibrahiminfinite, 2023-08-15:
@sjahr , 
This can be closed.
The keyboard and joystick control code are given only as a template for people to build on.

Maybe we can make a new issue about making the controller code more configurable soon.

sjahr, 2023-08-15:
I close this one. Do you mind creating the follow up issue @ibrahiminfinite

closed 2023-08-15 by sjahr

ibrahiminfinite, 2023-08-15:
> I close this one. Do you mind creating the follow up issue @ibrahiminfinite

Done