moveit/moveit2 #1988: MoveIt2 Humble Source Build Not Working

cfrauzem94, 2023-03-04:
### Description
When trying to build colcon workspace in this tutorial (https://moveit.ros.org/install-moveit2/source/)
>> colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release

Get this error

c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [kinematics_metrics/CMakeFiles/moveit_kinematics_metrics.dir/build.make:76: kinematics_metrics/CMakeFiles/moveit_kinematics_metrics.dir/src/kinematics_metrics.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1225: kinematics_metrics/CMakeFiles/moveit_kinematics_metrics.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
gmake[2]: *** [robot_trajectory/CMakeFiles/test_robot_trajectory.dir/build.make:76: robot_trajectory/CMakeFiles/test_robot_trajectory.dir/test/test_robot_trajectory.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1856: robot_trajectory/CMakeFiles/test_robot_trajectory.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [2min 32s, exited with code 2]

Summary: 19 packages finished [3min 50s]
  1 package failed: moveit_core
  4 packages had stderr output: controller_manager moveit_configs_utils moveit_core ros2controlcli
  34 packages not processed

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build? Source
* If binary, which release version?
* If source, which branch?
* Which RMW (Fast DDS or Cyclone DDS)? Using Cyclone DDS

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Should build

### Actual behaviour
Keeps throwing fatal error

### Backtrace or Console output
I have followed the ROS humble install tutorial to a dot
I have also tried installing MoveIt from binaries instead and but then the tutorial cannot finde the movit2_tutorials package

Also, at this point I am trying to figure out how to best get started with moveit2. Can I just use the docker and go from there or will that limit me eventually?

ibrahiminfinite, 2023-03-05:
Hi @cfrauzem94 ,

I recently came across this issue.
From what I understood, this  happens due to running out of RAM. The manual  also recommends 32 GB
> This build command will likely take a long time (20+ minutes) depending on your computer speed and amount of RAM available (we recommend 32 GB).

You should be able to get around  this by setting `--parallel-workers 1`
> colcon build --mixin release --parallel-workers 1

Let me know if this helps.

cfrauzem94, 2023-03-05:
Hi @ibrahiminfinite,

Thanks for your quick reply! 
Since I only have 16GB of memory, I added that setting the first time I had a different error. That did help the build go a bit longer but then I ran into the issue I posted.

I have not however tried

colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release -- parallel-workers 1

I am unsure what the difference is between

 --cmake-args -DCMAKE_BUILD_TYPE=Release 

 and

-- mixin release

Thanks,
Christoph

ibrahiminfinite, 2023-03-05:
Hi @cfrauzem94 ,

If setting the parallel workers to 1 did not help you.
You can try the solution mentioned here .
https://answers.ros.org/question/368106/c-fatal-moveit-2-error-killed-signal-terminated-program-cc1plus-compilation/

Another alternative is to increase create a larger swap space if you are using a docker or VM like WSL.  
And then build again.

To create and activate a swap space , you can follow this guide 
https://docs.rackspace.com/support/how-to/create-remove-swap-file-in-ubuntu/

tylerjw, 2023-03-10:
Please respond if the above did help you. If not, please re-open this issue and let us know how it is failing now.

closed 2023-03-10 by tylerjw