moveit/moveit2 #1991: linker failure related to Boost

bigrobinson, 2023-03-06:
Failing to build from source due to linker error related to Boost libraries.

### Your environment
* ROS Distro: Humble
* OS Version: Ubuntu 22.04
* Source build
* branch = main
* RMW = Cyclone DDS

### Steps to reproduce
Follow source build instructions to the letter (both MoveIt and Humble Moveit instructions) and execute build.

### Expected behaviour
All packages should build.

### Actual behaviour
Build of moveit_core fails.

### Backtrace or Console output
... ~/ws_moveit2$ colcon build --mixin release
Starting >>> moveit_common
Starting >>> srdfdom
Starting >>> moveit_resources_panda_description
Starting >>> launch_param_builder
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_task_constructor_msgs
Starting >>> moveit_resources_pr2_description
Starting >>> moveit_resources_prbt_support                                                                                                                                                          
Starting >>> rosparam_shortcuts
--- stderr: launch_param_builder                                                                                                                                          
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< launch_param_builder [1.25s]
Finished <<< moveit_common [1.48s]                                                                                                                                                              
Starting >>> rviz_marker_tools
Finished <<< moveit_resources_panda_description [2.09s]                                                                                                            
Starting >>> moveit_resources_panda_moveit_config
Finished <<< moveit_resources_prbt_support [2.42s]                                                                                                           
Finished <<< moveit_resources_fanuc_description [2.48s]
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_resources_pr2_description [2.49s]
Finished <<< moveit_resources_panda_moveit_config [2.12s]                                                                                                                           
Finished <<< moveit_resources_fanuc_moveit_config [2.03s]                                                                                                                           
Starting >>> moveit_resources
Finished <<< moveit_resources [2.14s]                                                                                                                                                                     
Finished <<< rviz_marker_tools [10.3s]                                                                                                                                                        
Finished <<< srdfdom [14.0s]                                                                                                                             
Starting >>> moveit_core
Starting >>> moveit_configs_utils
--- stderr: moveit_configs_utils                                                                                                                                                     
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
---
Finished <<< moveit_configs_utils [1.28s]
Finished <<< rosparam_shortcuts [18.7s]                                                                                                                  
Finished <<< moveit_task_constructor_msgs [21.1s]                                                             
[Processing: moveit_core]                               
--- stderr: moveit_core                                     
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::put_mem_block(void*)'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::raise_runtime_error(std::runtime_error const&)'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::get_default_error_string(boost::regex_constants::error_type)'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::get_mem_block()'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::verify_options(unsigned int, boost::regex_constants::_match_flags)'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
/usr/bin/ld: ../utils/libmoveit_test_utils.so.2.5.4: undefined reference to `boost::re_detail_106800::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
collect2: error: ld returned 1 exit status
gmake[2]: *** [robot_model/CMakeFiles/test_robot_model.dir/build.make:273: robot_model/test_robot_model] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1594: robot_model/CMakeFiles/test_robot_model.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< moveit_core [46.0s, exited with code 2]

Summary: 14 packages finished [1min 0s]
  1 package failed: moveit_core
  3 packages had stderr output: launch_param_builder moveit_configs_utils moveit_core
  40 packages not processed

tylerjw, 2023-03-10:
I'm sorry you are having trouble building moveit. I'm unable to reproduce this failure locally. Would you mind providing any additional information you might think is relevant? I noticed you are using gmake. Maybe you can try installing regular make to see if that matters? Another thing would be to make sure you don't have versions of boost installed from somewhere other than the official Ubuntu jammy repos.

JafarAbdi, 2023-04-12:
Closing due to inactivity. If you still encounter this issue, please reopen with additional information

tylerjw, 2023-05-02:
Are there more details you could provide? Se haven't seen similar issues in CI or locally.

drewskoots, 2023-05-05:
After some scattershot debugging it seems I had to edit the line [here](https://github.com/ros-planning/moveit2/blob/72dbc0067e7aab7aa5d9e040ccc4a2f0144ca9ef/moveit_ros/planning/planning_components_tools/CMakeLists.txt#L31) by adding ${Boost_LIBRARIES} to the end. I'm not sure if this is the cleanest fix, or why exactly this only happened to me in my docker container if its not appearing in the CI.

moriarty, 2023-08-15:
It's possible that something isn't exporting it's dependency on boost correctly, but yes strange that this isn't appearing in CI.

```
moveit_ros/planning/plan_execution/src/plan_execution.cpp:#include <boost/algorithm/string/join.hpp>
moveit_ros/planning/constraint_sampler_manager_loader/src/constraint_sampler_manager_loader.cpp:#include <boost/tokenizer.hpp>
moveit_ros/planning/constraint_sampler_manager_loader/src/constraint_sampler_manager_loader.cpp:      boost::char_separator<char> sep(" ");
moveit_ros/planning/constraint_sampler_manager_loader/src/constraint_sampler_manager_loader.cpp:      boost::tokenizer<boost::char_separator<char>> tok(constraint_samplers, sep);
moveit_ros/planning/constraint_sampler_manager_loader/src/constraint_sampler_manager_loader.cpp:      for (boost::tokenizer<boost::char_separator<char>>::iterator beg = tok.begin(); beg != tok.end(); ++beg)
moveit_ros/planning/planning_request_adapter_plugins/src/fix_start_state_bounds.cpp:#include <boost/math/constants/constants.hpp>
moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp:#include <boost/tokenizer.hpp>
moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp:    boost::char_separator<char> sep(" ");
moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp:    boost::tokenizer<boost::char_separator<char>> tok(adapters, sep);
moveit_ros/planning/planning_pipeline/src/planning_pipeline.cpp:    for (boost::tokenizer<boost::char_separator<char>>::iterator beg = tok.begin(); beg != tok.end(); ++beg)
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:#include <boost/program_options/parsers.hpp>
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:#include <boost/program_options/variables_map.hpp>
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  boost::program_options::options_description desc;
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  desc.add_options()("nthreads", boost::program_options::value<unsigned int>(&nthreads)->default_value(nthreads),
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:      "trials", boost::program_options::value<unsigned int>(&trials)->default_value(trials),
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  boost::program_options::variables_map vm;
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  boost::program_options::parsed_options po = boost::program_options::parse_command_line(argc, argv, desc);
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  boost::program_options::store(po, vm);
moveit_ros/planning/planning_components_tools/src/evaluate_collision_checking_speed.cpp:  boost::program_options::notify(vm);

```

yashi, 2023-09-10:
@drewskoots, I have the same problem building moveit2.  And I have the same fix on my local files.  But as @moriarty posted it seems the CI is building it without any problem.  I still can't figure out why.

tmayoff, 2023-10-17:
I have also encountered this issues, I'm running Debian 12, and the above fix worked, however I also needed to add it to the `moveit_warehouse_*` libraries as well

moriarty, 2024-02-07:
I was unable to reproduce this locally on 22.04 ...

I'm going to close this issue, to reopen it we would need a docker image where we can reproduce it

closed 2024-02-07 by moriarty