moveit/moveit2 #1993: Cartesian Planning with TCP Velocity

ClaBirk, 2023-03-08:
# 6DOF Planner for cartesian movement with speed constrain (m/s) and orientation constrain (TCP)
looking for a Motion planner wich is able to control the **linear movement** of 6DOF Robot with a given speed in m/s. And detection if the set speed exceeds the joint limits. 
I couldnt find this feature or a demo for linear movement with variable speed.

henningkayser, 2023-03-09:
@ClaBirk the PILZ industrial planner is available. Please take a look at the tutorials, cartesian limits are documented here https://moveit.picknik.ai/humble/doc/examples/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#cartesian-limits

tylerjw, 2023-03-10:
Please re-open this issue if the above link doesn't answer your questions.

closed 2023-03-10 by tylerjw