moveit/moveit2 #2014: Testing MoveIt2 Python Bindings with real UR robot

rlee3359, 2023-03-14:
Hello! I'm attempting to use the new (and greatly appreciated) python bindings to control a real UR robot, and I was wondering if I could get some guidance on setting things up correctly. 

I'm having some trouble following how to apply the tutorial information [here](https://github.com/ros-planning/moveit2_tutorials/tree/main/doc/examples/motion_planning_python_api) to the UR. I've got this python panda example running fine.

I've also set up the [UR drivers](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) and got the robot working with MoveIt in the rviz plugin, using this [moveit launch file](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/main/ur_moveit_config/launch/ur_moveit.launch.py). I would like to now adapt this [launchfile](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/motion_planning_python_api/launch/motion_planning_python_api_tutorial.launch.py) to handle the UR robot.

The tutorial mentions needing a yaml file for the planners, which I have taken from the example, but there seems to be more going on with the moveit config builder parameters being passed as a dictionary to the script. Is there an easy way to set this up for the UR? The aforementioned moveit config launchfile from the UR drivers package includes the robot description and some controller info, can I pull it from there or perhaps combine the launchfiles? 

Another question is if I make a python script and run it with`ros2 run`, how can I pass all the required params? 

Any help with this would be greatly appreciated!

henningkayser, 2023-03-15:
@rlee3359 great that you are testing this new API. The tutorials are still being worked on, so it's possible that there are already fixes in the moveit2_tutorials [PR list](https://github.com/ros-planning/moveit2_tutorials/pulls).

Maybe @peterdavidfagan has a suggestion on your launch setup.
In general, it won't be possible to only use `ros2 run`, since the launch setup is taking care about all the parameters and other required nodes like driver and controllers.

JafarAbdi, 2023-04-12:
@rlee3359 Do you mind sharing your moveit_configs package? I've been able to test the python binding with multiple robots (including UR) without an issue. I'm away from my work machine for the next two weeks, so I can't share it

cpr-fer, 2023-04-20:
Hi @peterdavidfagan

I also really appreciate the new API, thanks a lot for implementing it!

> You may want to start looking into the [MoveItConfigsBuilder](https://github.com/ros-planning/moveit2/blob/main/moveit_configs_utils/moveit_configs_utils/moveit_configs_builder.py), it greatly simplifies the launchfile.

Is this a requirement or just to make the yaml->dict conversation easier? I am struggeling right now with the same issue of loading the .yaml file, just for igus robots instead of ur, since my ROS launch file can't easily switch to MoveItConfigsBuilder.

peterdavidfagan, 2023-04-20:
Hi @cpr-fer, 

It is not a requirement but it does make loading parameters a lot easier. What issue are you encountering in using the MoveItConfigsBuilder in your launchfile?

cpr-fer, 2023-04-20:
MoveItPy does not find the pipeline parameters when loading the yaml file via:
``` python
# launch file
    moveitpy_yaml = os.path.join(
        get_package_share_directory("irc_ros_moveit_config"),
        "config",
        "moveit_py.yaml", # Pretty much the same file as used in the examples
    )
    moveitpy = load_yaml(Path(moveitpy_yaml))

   [...]

    baristabot_node = Node(
        name="moveit_py",
        package="irc_ros_examples",
        executable="baristabot.py",
        namespace=namespace,
        parameters=[
            [...]
            moveitpy,
        ],
    )

```

And 
``` python
# Node
    node = Node("param_checker")
    node.declare_parameter("planning_pipelines.pipeline_names")
    out = node.get_parameter("planning_pipelines.pipeline_names")
    node.get_logger().error(str(out.value))
    rclpy.shutdown()
    # Returns 
    # [baristabot.py-6] [ERROR] [1681993872.796709776] [moveit_py]: ['ompl', 'pilz_industrial_motion_planner']

    moveit_py_instance = MoveItPy(node_name="moveit_py")
    # Returns
    # [baristabot.py-6] [ERROR] [1681993872.966974646] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any planning pipelines.
    # [baristabot.py-6] [FATAL] [1681993872.966995184] [moveit.ros_planning_interface.moveit_cpp]: Failed to load planning pipelines from parameter server
```

The entire code can be found [here](https://github.com/CommonplaceRobotics/iRC_ROS/blob/19-implement-baristabot-example/irc_ros_examples/scripts/baristabot.py) but is very much WIP. Also I am unsure if this shouldn't be a seperate issue, I don't want to hijack this one with my (related?) problems

peterdavidfagan, 2023-04-20:
On quick inspection what you are doing looks fine and should work. I would need to to reproduce the error on my machine and debug from there. 

We read the parameters generated by the launch script before spinning a node. The utility to get these parameters is [here](https://github.com/ros-planning/moveit2/blob/f0debfd1036c13ac06194a17e400a95616325cf5/moveit_py/moveit/utils.py#L50), an error is likely due to this utility. In the bindings you can see we read the parameters in the following [section](https://github.com/ros-planning/moveit2/blob/f0debfd1036c13ac06194a17e400a95616325cf5/moveit_py/src/moveit/moveit_ros/moveit_cpp/moveit_cpp.cpp#L70).

I won't be able to look in more depth today but I plan to work on a review for an open pr this week so I can group this with that work in case there is a bug with the library. Thank you for sharing the above.

cpr-fer, 2023-04-21:
@peterdavidfagan thank you! If you're not faster than me I'll try to test a similiar launch file setup on Sunday with the UR drivers to ensure it is not an issue with my robot driver code

cpr-fer, 2023-04-25:
Hi @peterdavidfagan

I printed the parameter dicts from the moveit tutorial and the UR driver I linked above, loaded them into Python and checked the planning pipelines parameters, which seem to be identical:
```
>>> ur["planning_pipelines"]
{'pipeline_names': ['ompl', 'pilz_industrial_motion_planner', 'chomp', 'ompl_rrt_star']}
>>> moveit["planning_pipelines"]
{'pipeline_names': ['ompl', 'pilz_industrial_motion_planner', 'chomp', 'ompl_rrt_star']}
```
Still only the ur launchfile complains about not finding that parameter

I'll make sure the rest of the parameters except robot descriptions are identical in the afternoon

cpr-fer, 2023-04-25:
> The tutorial examples are specifically for the panda robot. If you are adapting these to the UR this makes sense why there are errors.

Ye that was what I am doing, I used [this](https://github.com/ros-planning/moveit2_tutorials/blob/main/doc/examples/motion_planning_python_api/scripts/motion_planning_python_api_tutorial.py) file as I didnt wan't to delve into the UR stuff to much just now and was happy to get the node to start.

> If you do run the library with your UR, you will have been the first to do so, which would be so awesome. Video of the robot operating using the library would be much appreciated.

Sadly I don't have a real UR so I can only provide a mock/simulated UR once I get to change the tutorial file sometime later this week. The main goal for me is to get it running on my igus ReBeL first, after that I'll look into the UR :)

hello-chintan, 2023-05-02:
@peterdavidfagan, excellent work on creating the Python bindings for MoveIt 2! Thanks! I was able to get them working on a Hello Robot Stretch 2. The launch config setup is a bit convoluted because the [stretch_moveit2](https://github.com/hello-robot/stretch_ros2/tree/rolling/stretch_moveit2) package is not organized correctly to use MoveItConfigsBuilder, but if anyone is looking for an example setup, this could be helpful.

I'll try to clean it up more so things are more straightforward. I would also appreciate any pointers for an easier setup as right now all the required parameters are loaded from a single YAML called [motion_planning_python.yaml](https://github.com/hello-robot/stretch_ros2/blob/rolling/stretch_moveit2/config/motion_planning_python.yaml).

peterdavidfagan, 2023-05-02:
> I would also appreciate any pointers for an easier setup as right now all the required parameters are loaded from a single YAML called [motion_planning_python.yaml](https://github.com/hello-robot/stretch_ros2/blob/rolling/stretch_moveit2/config/motion_planning_python.yaml).

It hasn't been fully ported to ROS 2 yet but the MoveIt Setup Assistant can really help in generating a MoveIt Config packages. Example packages can be found [here](https://github.com/ros-planning/moveit_resources/tree/ros2/panda_moveit_config). I have a package for the Franka Emika Panda that I would also be happy to share.

tylerjw, 2023-05-02:
In case anyone wants to try it I just built the main branch on humble into debains here: https://github.com/moveit/moveit2_packages/tree/jammy-humble

To try it run these commands to add it to your apt list and then you should be able to install moveit-py in the normal ros way through apt:

```shell
echo "deb [trusted=yes] https://raw.githubusercontent.com/moveit/moveit2_packages/jammy-humble/ ./" | sudo tee /etc/apt/sources.list.d/moveit_moveit2_packages.list
echo "yaml https://raw.githubusercontent.com/moveit/moveit2_packages/jammy-humble/local.yaml humble" | sudo tee /etc/ros/rosdep/sources.list.d/1-moveit_moveit2_packages.list
```

cpr-fer, 2023-05-03:
> I'll try to clean it up more so things are more straightforward. I would also appreciate any pointers for an easier setup as right now all the required parameters are loaded from a single YAML [...]

Hi @hello-chintan
with the help of OpaqueFunction and merging everything into a single dictionary at the end I got the loading of multiple config files for moveitpy to work [here](https://github.com/CommonplaceRobotics/iRC_ROS/blob/humble/irc_ros_moveit_config/launch/rebel.launch.py). Note that it is still very messy and I need to reduce the number of files which are loaded at some point

hello-chintan, 2023-06-27:
> In case anyone wants to try it I just built the main branch on humble into debains here: https://github.com/moveit/moveit2_packages/tree/jammy-humble

Hey @tylerjw,
I just tried installing moveit-py using the humble binaries, but importing the MoveItPy class results in the following error:
```
---------------------------------------------------------------------------
ImportError                               Traceback (most recent call last)
<ipython-input-1-d55c8c99cba3> in <module>
----> 1 from moveit.planning import MoveItPy

/opt/ros/humble/local/lib/python3.10/dist-packages/moveit/__init__.py in <module>
----> 1 from moveit.core import *
      2 from moveit.planning import *
      3 from moveit.utils import get_launch_params_filepath

ImportError: libmoveit_planning_scene.so.2.7.4: cannot open shared object file: No such file or directory

```

Any tips on how I could get around this?

hello-chintan, 2023-06-27:
Ah, I forgot to update the rest of the moveit packages through apt after installing the moveit-py package. It works now! Thanks!

ZealousGinger, 2023-07-03:
Hey Im also having similar issues, on Humble building from source in main. Im using the MoveItConfigsBuilder and not sure what went wrong.
```
[ERROR] [1688372176.747794913] [moveit.ros_planning_interface.moveit_cpp]: Failed to load any planning pipelines.
[FATAL] [1688372176.747824181] [moveit.ros_planning_interface.moveit_cpp]: Failed to load planning pipelines from parameter server

```
Does anyone know any possible solution to solve this issue? Thank you!

closed 2023-08-24 by tylerjw

Mennosytsma, 2023-10-26:
I see that this issue is closed, but the proposed solutions combined with my basic knowledge make that I still cannot reach the initial goal the title suggests. 
Based on the previous discussions I wanted to proceed with the files @cpr-fer linked to, which resulted in the package [here](https://github.com/Mennosytsma/moveitpy_pkg/tree/main).
I have changed the joint names in `motion_planning_python_api_tutorial.py` to match the UR (the values in the python script are still referencing to panda, but if it gets the right descriptions that should not be an issue, hope to fix that later)
I also preferred the method of @cpr-fer to make only minor changes to the original `ur_moveit.launch.py`, but I think this results in not enough information in the moveit_config dict, as I get the following error when it runs `plan_result = planning_component.plan()` 

```
[WARN] [1698332153.851265890] [moveit.ompl_planning.planning_context_manager]: Cannot find planning configuration for group 'ur_manipulator' using planner 'RRTConnectkConfigDefault'. Will use defaults instead.
[WARN] [1698332153.851414927] [moveit.ompl_planning.model_based_planning_context]: It looks like the planning volume was not specified.
[INFO] [1698332153.851589965] [moveit.ompl_planning.model_based_planning_context]: Planner configuration 'ur_manipulator' will use planner 'geometric::RRTConnect'. Additional configuration parameters will be set when the planner is constructed.
[WARN] [1698332153.903266767] [moveit.ros_planning.planning_pipeline]: The planner plugin did not fill out the 'planner_id' field of the MotionPlanResponse. Setting it to the planner ID name of the MotionPlanRequest assuming that the planner plugin does warn you if it does not use the requested planner.
```
The resulting robot trajectory has a duration of 0, which indicated to me that something went wrong in the planning, but it is not clear to me what more parameters I need to add to the `moveit_py`  node to get the interface to work correctly. 

**Testing environment**
To test this code I make use of Ubuntu 22, ROS2 Rolling and [UR ros2 driver](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver) and [description](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description)

Next to running test_moveitpy.launch with the argument ur_type:=ur5 I run the default `ur_control`  launch file in a separate terminal and have a ur5 simulated in [docker](https://hub.docker.com/r/universalrobots/ursim_cb3) that runs external control, dragging and planning a path using RVIZ works perfectly fine, but the tutorial does not work yet

adamanov, 2024-06-14:
> Hi @adamanov,
> 
> Apologies for delay in response I've be off work sick due to a virus.
> 
> Can you share your planner config file. Here is an example of one I use: https://github.com/peterdavidfagan/ros2_robotics_research_toolkit/blob/franka_emika_panda/src/applications/motion_planning/config/moveit_cpp.yaml.
> 
> I have launch files in: https://github.com/peterdavidfagan/ros2_robotics_research_toolkit. These files are found under the control_launch_configs and application folders of this workspace.
> 
> These are functional but are for the Franka Emika Panda, also there is some separation between launching controllers and motion planning prerequisites so it might not be the easiest to get started with. I am happy to help with any specific questions you have.

thanks for the help, I got it to work :)) I was missing to parse the parameter of MoveItConfigsBuilder into the Node where MoveitPy is started. 
But it is not intuitive that MoveItPy is not grasping the parameters directly from parameter server  when the move_group_node is already launched and got the parameters 

For future folks:
1. launch move_group_node with all necessary parameters, controllers etc ...
2. launch the MoveitPy node
i.e. 

```
    moveit_config = (
        MoveItConfigsBuilder(
            robot_name="omp_gripper", package_name="omp_gripper_no_jackal_moveit_config"
        )
        .robot_description(
            file_path="config/open_manipulator_p.urdf.xacro",
            mappings={
                "ros2_control_hardware_type": LaunchConfiguration(
                    "ros2_control_hardware_type"
                )
            },
        )
        .robot_description_semantic(file_path="config/open_manipulator_p.srdf")
        .planning_scene_monitor(
            publish_robot_description=True, publish_robot_description_semantic=True
        )
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .joint_limits(file_path="config/joint_limits.yaml")
        .moveit_cpp(file_path="config/moveit_cpp.yaml")
        .to_moveit_configs()
    )

    moveit_config_dict = moveit_config.to_dict()
    moveit_py_node = Node(
        name="moveit_py",
        package="panda_motion_planning_demos",
        executable="motion_planning_omp.py",
        output="both",
        arguments=["--ros-args", "--log-level", "info"],
        parameters=[
            moveit_config_dict,
        ],
    )
   
 ```

will-d-chen, 2024-11-11:
I've managed to get moveit working in ROS2 with ur_robot_driver and have some examples in python. I made a pretty comprehensive tutorial, hopefully this helps anyone who is still confused. https://github.com/will-d-chen/ur3e-codebase