moveit/moveit2 #2018: For convenience, add a vector version of `RobotState::getVariableVelocities()`

AndyZe, 2023-03-14:
It would be convenient to add a std::vector-ized version of this:

```
  const double* getVariableVelocities() const
  {
    return velocity_;
  }
```

henningkayser, 2023-03-15:
There is already https://moveit.picknik.ai/humble/api/html/classmoveit_1_1core_1_1RobotState.html#a3c2938fb8c9467dff253db4b5c1992ca. But if you think the copy functions are missing for the all-variable getters, I'd say it would be consistent to add them

ibrahiminfinite, 2023-03-18:
Hi,

While going through [robot_state.h](https://github.com/ros-planning/moveit2/blob/main/moveit_core/robot_state/include/moveit/robot_state/robot_state.h) .
 I noticed that it contains a good number of definitions.
At first I though this was mostly for `const` return functions. but then realized that was not the case.

Is there a reason why some functions like `setVariableAccelerations()` has one of the overloads defined in the `robot_state.h` while the other overloads are in the `robot_state.cpp` ?

Tejal-19, 2023-03-18:
Hey, I have tried doing the same and created a PR #2023. Please let me know if anything else is required.

AndyZe, 2023-03-19:
@ibrahiminfinite that's a good point! I think you should open another issue about that and file a PR about it if you have time. I'd call it something like "Consistent function definitions in RobotState"

As a rule of thumb for robot_state.h, perhaps we can say:

- If a function definition is 2 lines or less, put it in `robot_state.h`
- If a function definition is >2 lines, put it in `robot_state.cpp`

I'm curious what @henningkayser and @tylerjw think about this rule of thumb

ibrahiminfinite, 2023-03-19:
I am not a fan of definitions in the header,  unless it is a single line return.
But if the rule of thumb you stated is the way things are done within moveit ,  we can go with that.

Tejal-19, 2023-04-01:
@AndyZe do you want me to make any changes to the PR ?

AndyZe, 2023-04-01:
@Tejal-19 let's wait for the response to this comment:  https://github.com/ros-planning/moveit2/pull/2023#issuecomment-1475089319

ibrahiminfinite, 2023-08-15:
I also think it would be nice to have a  method  that returns a vector rather than copy it into one.
Especially useful in places where we don't  just need use the result without storing.

We can close this if such a method won't be of much use.

closed 2024-03-15 by github-actions[bot]