moveit/moveit2 #2036: How to Efficiently Implement Dual Arm Planning using computeCartesianPath

yeyanlei, 2023-03-22:
### Description

Overview of your issue here.
I am working on a project involving dual arm planning, and I am trying to set up the arm_group to include both right_arm_group and left_arm_group. I would like to know how to use computeCartesianPath to plan for the arm_group that includes both arms.
### Your environment
* ROS Distro: [Foxy|Humble|Rolling]
* OS Version: e.g. Ubuntu 22.04
* Source or Binary build?
* If binary, which release version?
* If source, which branch?
* Which RMW (Fast DDS or Cyclone DDS)? Humble or Rolling should be using Cyclone.

### Steps to reproduce
Tell us how to reproduce this issue. Attempt to provide a working demo, perhaps using Docker.

### Expected behaviour
Tell us what should happen

### Actual behaviour
Tell us what happens instead

### Backtrace or Console output

Use [gist.github.com](gist.github.com) to copy-paste the console output or segfault backtrace using gdb.

yeyanlei, 2023-03-23:
@felixvd @henningkayser Could you give me some help. Thank you very much.

henningkayser, 2023-03-23:
There is currently no support for multiple end-effectors in `computeCartesianPath()`. It might be possible to extend the CartesianInterpolator API to support multiple groups, or simply to run two Cartesian paths and check for trajectory collisions afterwards. Same can be tried using PILZ's LIN command (https://moveit.picknik.ai/humble/doc/examples/pilz_industrial_motion_planner/pilz_industrial_motion_planner.html#the-lin-motion-command). Currently. the cleanes approach would be to feed both Cartesian planning requests into an [MTC parallel container](https://moveit.picknik.ai/humble/doc/examples/moveit_task_constructor/moveit_task_constructor_tutorial.html), but also there I'm not certain about the robustness of multi-trajectory collision checking.

yeyanlei, 2023-03-24:
@henningkayser , Thank you very much for your reply. I appreciate your insight on this issue. I will look into extending the CartesianInterpolator API to support multiple groups or using LIN command in PILZ as you suggested. I will also consider using MTC parallel container for both Cartesian planning requests. Thank you again for your help.

closed 2023-04-03 by yeyanlei