moveit/moveit2 #2052: Force-based Stopping During Trajectory Execution

emfebert, 2023-03-27:
Hi,

I'd like to add a functionality that allows terminating a trajectory when the force sensed by a 6-axis force sensor mounted at the end-effector exceeds a specified threshold.

Where would be the best location to add this? Inside the controller perhaps?

Thanks a lot!

JafarAbdi, 2023-04-12:
Which interface are you using MoveItCpp or MoveGroup? 

You do have access to the trajectory execution manager for the former, you could have a subscriber to the sensor and call [stopExectution](https://moveit.picknik.ai/main/api/html/classtrajectory__execution__manager_1_1TrajectoryExecutionManager.html#a1fb8ee6f7c809152afc3989dc7447046) when you exceed the specified threshold. 

For MoveGroup you could create a publisher with string message to the following topic [EXECUTION_EVENT_TOPIC](https://moveit.picknik.ai/main/api/html/classtrajectory__execution__manager_1_1TrajectoryExecutionManager.html#a93ea7dd3bead6b8422e231b51cb165ce) and send "stop" to stop the execution (or cancel through the action interface)

sjahr, 2023-07-13:
Closing this due to inactivity. @emfebert Feel free to re-open if you still need an answer to this question

closed 2023-07-13 by sjahr