moveit/moveit2 #2053: Unable to locate package ros-rolling-moveit

joao-pm-santos96, 2023-03-28:
### Description

I am trying to install ros-rolling-moveit but my package manager is not able to find it. In fact, when accessing http://packages.ros.org/ros2/ubuntu/ we can check that the last update was some days ago... can we predict when the packages will be avaiable?

### Your environment
* ROS Distro: Rolling
* OS Version: Ubuntu 22.04
* Binary build

### Steps to reproduce
When running `sudo apt update && sudo apt instal ros-rolling-moveit` the output is `E: Unable to locate package ros-rolling-moveit` but I am able to install `moveit-common`, `moveit-msgs` and others.

gavanderhoorn, 2023-03-28:
Please see [New packages for ROS 2 Rolling Ridley 2023-03-17](https://discourse.ros.org/t/new-packages-for-ros-2-rolling-ridley-2023-03-17/30365/2) on Discourse.

PS: this is also not a "bug" I believe.

joao-pm-santos96, 2023-03-28:
Thank you for the tips! Indeed not a bug, just a lack of knowledge. Will close it then.

closed 2023-03-28 by joao-pm-santos96