moveit/moveit2 #2056: Docker humble-source has rolling source code

MarcoMagriDev, 2023-03-30:
### Description

`moveit/moveit2:humble-source` docker image has `main` branch code.

### Your environment
* ROS Distro: Humble
* OS Version: e.g. Ubuntu 22.04
* `moveit/moveit2:humble-source` docker image

### Steps to reproduce
1) Pull the image `docker pull moveit/moveit2:humble-source`
2) Run the container and go to the `~/ws_moveit/src/moveit2/` folder
3) Run `git log` to check the last commit hash

### Expected behaviour
The last commit should be 259f039c9a4e5ce55d3f3593246101040c279b59 since at the current time it is the last commit on the `humble` github branch. 

### Actual behaviour
The last commit (9bf6ae34197b9a64fd8c915f49b13deaaf726d23 in my case) belongs to the main branch instead of the humble one.

tylerjw, 2023-03-30:
I think this is intentional. The release branches are meant to track what we release into ros. If you want what is released into ros you can just use the binaries. The docker containers provide a way to get pre-built versions of moveit on main.

MarcoMagriDev, 2023-03-31:
> I think this is intentional. The release branches are meant to track what we release into ros. If you want what is released into ros you can just use the binaries. The docker containers provide a way to get pre-built versions of moveit on main.

But since they are using the same source code what is the difference between the `moveit/moveit2:humble-source` and the `moveit/moveit2:rolling-source` images?

vatanaksoytezer, 2023-04-04:
> But since they are using the same source code what is the difference between the `moveit/moveit2:humble-source` and the `moveit/moveit2:rolling-source` images?

It's the same code, one runs on rolling one runs on humble, that's the only difference for now.

vatanaksoytezer, 2023-04-05:
@MarcoMagriDev I am going to close this issue as answered, but feel free to ask further clarifying questions, re-open the issue or open new issues if you encounter any problems.

closed 2023-04-05 by vatanaksoytezer

Zarnack, 2024-02-26:
Hey,
it seems like this has changed? If I am currently pulling moveit/moveit2:humble-source I am getting the humble branch. 
Will this change in the future again? Currently I am not able to use moveit_py with the humble-source image anymore as it is not available on the humble branch. 
Is it maybe possible to have 2 humble-source images? One tracking the humble branch and one tracking the main branch (to have all recent features available)?